timvideos/litex-buildenv #2: Need to set UTF-8 environment to complete

duncan-roe, 2018-01-05:
With just the C locale, when installing openocd we get the error "UnicodeEncodeError: 'ascii' codec can't encode characters in position 118-119: ordinal not in range(128)".
The build instructions need to advise to set UTF-8 environment

TomKeddie, 2019-05-19:
I'm in Canada so I used export LANG="en_CA.utf8".  In the US you would use export LANG="en_US.utf8" etc.

```tom@z400:~/tmp/litex$ locale -a
C
en_AU
en_AU.iso88591
en_AU.utf8
en_CA
en_CA.iso88591
en_CA.utf8
en_US
en_US.iso88591
en_US.utf8
POSIX
tom@z400:~/tmp/litex$ locale
LANG=en_CA
LC_CTYPE="en_CA"
LC_NUMERIC="en_CA"
LC_TIME="en_CA"
LC_COLLATE="en_CA"
LC_MONETARY="en_CA"
LC_MESSAGES="en_CA"
LC_PAPER="en_CA"
LC_NAME="en_CA"
LC_ADDRESS="en_CA"
LC_TELEPHONE="en_CA"
LC_MEASUREMENT="en_CA"
LC_IDENTIFICATION="en_CA"
LC_ALL=
tom@z400:~/tmp/litex$ export LANG="en_CA.utf8"
tom@z400:~/tmp/litex$ locale
LANG=en_CA.utf8
LC_CTYPE="en_CA.utf8"
LC_NUMERIC="en_CA.utf8"
LC_TIME="en_CA.utf8"
LC_COLLATE="en_CA.utf8"
LC_MONETARY="en_CA.utf8"
LC_MESSAGES="en_CA.utf8"
LC_PAPER="en_CA.utf8"
LC_NAME="en_CA.utf8"
LC_ADDRESS="en_CA.utf8"
LC_TELEPHONE="en_CA.utf8"
LC_MEASUREMENT="en_CA.utf8"
LC_IDENTIFICATION="en_CA.utf8"
LC_ALL=
```

mithro, 2019-05-20:
@TomKeddie Can you give me the command that you were running when this error occurs?

TomKeddie, 2019-05-20:
@mithro I'm on archlinux but I'm not sure that's related.  Here you go
```
tom@z400:~/tmp/litex1$ curl -fsS https://raw.githubusercontent.com/timvideos/litex-buildenv/master/scripts/bootstrap.sh | bash
Distributor ID: Arch
bash: line 43: ./scripts/debian-setup.sh: No such file or directory
             This script is: /home/tom/tmp/litex1/HDMI2USB-litex-firmware/scripts/download-env.sh
         Firmware directory: /home/tom/tmp/litex1/HDMI2USB-litex-firmware
         Build directory is: /home/tom/tmp/litex1/HDMI2USB-litex-firmware/build
     3rd party directory is: /home/tom/tmp/litex1/HDMI2USB-litex-firmware/third_party

Initializing environment
---------------------------------

Installing conda (self contained Python environment with binary package support)
In /home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/lib/python3.6/site-packages conda/common/path.py is already patched.
Warning: 'timvideos' already in 'channels' list, moving to the top
.
.
.
.
.
Installing openocd (jtag tool for programming and debug)
Collecting package metadata: done
Solving environment: done

## Package Plan ##

  environment location: /home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda

  added / updated specs:
    - openocd=0.10.0

The following NEW packages will be INSTALLED:

  openocd            timvideos/linux-64::openocd-0.10.0_0807_gbbdb820c8-20190504_170606

Preparing transaction: done
Verifying transaction: failed

# >>>>>>>>>>>>>>>>>>>>>> ERROR REPORT <<<<<<<<<<<<<<<<<<<<<<

    Traceback (most recent call last):
      File "/home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/lib/python3.6/site-packages/conda/exceptions.py", line 1003, in __call__
        return func(*args, **kwargs)
      File "/home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/lib/python3.6/site-packages/conda/cli/main.py", line 84, in _main
        exit_code = do_call(args, p)
      File "/home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/lib/python3.6/site-packages/conda/cli/conda_argparse.py", line 82, in do_call
        exit_code = getattr(module, func_name)(args, parser)
      File "/home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/lib/python3.6/site-packages/conda/cli/main_install.py", line 20, in execute
        install(args, parser, 'install')
      File "/home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/lib/python3.6/site-packages/conda/cli/install.py", line 275, in install
        handle_txn(unlink_link_transaction, prefix, args, newenv)
      File "/home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/lib/python3.6/site-packages/conda/cli/install.py", line 304, in handle_txn
        unlink_link_transaction.execute()
      File "/home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/lib/python3.6/site-packages/conda/core/link.py", line 236, in execute
        self.verify()
      File "/home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/lib/python3.6/site-packages/conda/common/io.py", line 88, in decorated
        return f(*args, **kwds)
      File "/home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/lib/python3.6/site-packages/conda/core/link.py", line 223, in verify
        exceptions = self._verify(self.prefix_setups, self.prefix_action_groups)
      File "/home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/lib/python3.6/site-packages/conda/core/link.py", line 522, in _verify
        for target_prefix, prefix_group in iteritems(prefix_action_groups)),
      File "/home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/lib/python3.6/site-packages/conda/core/link.py", line 518, in <genexpr>
        exceptions = tuple(exc for exc in concatv(
      File "/home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/lib/python3.6/site-packages/conda/core/link.py", line 346, in _verify_individual_level
        error_result = axn.verify()
      File "/home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/lib/python3.6/site-packages/conda/core/path_actions.py", line 282, in verify
        if self.link_type != LinkType.directory and not lexists(self.source_full_path):  # pragma: no cover  # NOQA
      File "/home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/lib/python3.6/posixpath.py", line 179, in lexists
        os.lstat(path)
[... 25 lines cut ...]

     active environment : None
       user config file : /home/tom/.condarc
 populated config files : /home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/.condarc
          conda version : 4.6.14
    conda-build version : not installed
         python version : 3.6.0.final.0
       base environment : /home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda  (writable)
           channel URLs : https://conda.anaconda.org/timvideos/linux-64
                          https://conda.anaconda.org/timvideos/noarch
                          https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/free/linux-64
                          https://repo.anaconda.com/pkgs/free/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/pkgs
       envs directories : /home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/envs
                          /home/tom/tmp/litex1/HDMI2USB-litex-firmware/build/conda/.conda/envs
               platform : linux-64
             user-agent : conda/4.6.14 requests/2.21.0 CPython/3.6.0 Linux/5.0.9-arch1-1-ARCH arch/rolling glibc/2.29
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

mithro, 2019-05-20:
The issue seems to be that openocd has added two files which have unicode in them;
```
./share/openocd/scripts/target/к1879xб1я.cfg
./share/openocd/scripts/target/1986ве1т.cfg
```

mithro, 2019-05-21:
Might be fixed by https://github.com/timvideos/conda-hdmi2usb-packages/commit/c0a696bee8e4819cd614a6ec29bdd96fd721b18d when a new conda package is uploaded. Will test shortly....

mithro, 2019-05-21:
The new conda package has been uploaded, can you try again now?

TomKeddie, 2019-05-21:
> The new conda package has been uploaded, can you try again now?

Thanks Tim. This issue seems fixed in litex-buildenv now. 
@duncan-roe are you able to verify, you will need to pull from litex-buildenv, set the env var as below.

```export GITHUB_REPO=litex-buildenv```

duncan-roe, 2019-05-21:
Using your **export** command I was able to pull the repo. Oddly, **git pull** without it failed 

> fatal: repository 'https://github.com/nizox/tapcfg/' not found

after asking for and getting my login. Should I update _.git/config_?
I am sorry but it will be a while before I can test this. The issue arose at LCA2018 and I still have to figure out why one of the Lolibot wheels doesn't turn before I get back to it (Multimeter expired, keep meaning to get a new one ...)

duncan-roe, 2019-05-22:
Using your **export** command I was able to pull the repo. Oddly, **git pull** without it failed 

> fatal: repository 'https://github.com/nizox/tapcfg/' not found

after asking for and getting my login. Should I update _.git/config_?
I am sorry but it will be a while before I can test this. The issue arose at LCA2018 and I still have to figure out why one of the Lolibot wheels doesn't turn before I get back to it (Multimeter expired, keep meaning to get a new one ...)
Oops Litex is for the FPGA project. not Lolibot. That's behind Lolibot on my list atm

mithro, 2019-05-22:
https://github.com/nizox/tapcfg/ moved to https://github.com/enjoy-digital/tapcfg

Try doing a `git submodule sync --recursive`?

mithro, 2019-07-26:
I believe this has been fixed now.

closed 2019-07-26 by mithro