timvideos/litex-buildenv #6: Look at supporting tftp booting using a "high port" which can be opened by non-root users

mithro, 2018-01-16:
"Low" ports can only be opened by root (or user with special privileges) -- https://www.staldal.nu/tech/2007/10/31/why-can-only-root-listen-to-ports-below-1024/

tftp by default uses port 68 (https://en.wikipedia.org/wiki/Trivial_File_Transfer_Protocol) -- is there an alternative port?

mithro, 2018-01-16:
https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libnet/tftp.c#L7

mithro, 2018-01-16:
Python Libraries - 
 * https://wiki.python.org/moin/tftp
 * https://pypi.python.org/pypi/tftpy
 * https://github.com/sirMackk/py3tftp

ewen-naos-nz, 2018-01-18:
`atftpd` will take a `-p PORT` option to force it to listen on a different port; `in.tftpd` seems to have a `-R PORT:PORT` option to constrain the server port to be in that range which may be sufficient (seems to support multiple ports to allow simultaneous transfer from the same client).

`litex/soc/software/net/tftp.c` is trivially modified to take `server_port` as a parameter, which allows `litex/soc/software/bios/boot.c` to specify the TFTP port in a similar manner to the IP.  (The callback just replies to the port it received traffic from which is simple, if not very secure.)

Once `server_port` is under BIOS control it should be relatively simple to have it try a list of ports in turn, at least for the first file, until it finds one that works.

mithro, 2018-01-18:
@ewen-naos-nz - SGTM!

ewen-naos-nz, 2018-01-18:
@mithro: [litex pull request 52 - BIOS: Support alternative TFTP server port](https://github.com/enjoy-digital/litex/pull/52).

When compiled with `TFTP_SERVER_PORT=6969` exported in the environment, it will try both UDP/6969 and then UDP/69 (for `boot.bin`) during boot.  Eg,

    (LX P=arty C=or1k.linux F=linux) ewen@parthenon:/src/fpga-miniconf/litex-buildenv$ make firmware-connect
    flterm --port=/dev/ttyUSB1 --speed=115200
    [FLTERM] Starting...

    LiteX SoC BIOS (or1k)
    (c) Copyright 2012-2018 Enjoy-Digital
    (c) Copyright 2007-2018 M-Labs Limited
    Built Jan 18 2018 13:11:19

    BIOS CRC passed (189f291e)
    Initializing SDRAM...
    Memtest OK
    Booting from serial...
    Press Q or ESC to abort boot completely.
    sL5DdSMmkekro
    Timeout
    Booting from network...
    Local IP : 192.168.100.50
    Remote IP: 192.168.100.100
    Fetching from: UDP/6969
    Jan 18 13:23:40 parthenon atftpd[10761.139727563720448]: socket may listen on any address, including broadcast
    Jan 18 13:23:40 parthenon atftpd[10761.139727563720448]: Serving boot.bin to 192.168.100.50:7642
    Unable to download boot.bin over TFTP
    Fetching from: UDP/69
    [....]
    Successfully downloaded 3730908 bytes from boot.bin over TFTP

We still need to, eg, support starting TFTP server on alternative port.

ewen-naos-nz, 2018-01-18:
FTR: once `enjoy-digital/litex` merges pull request, we'll also have to update our `git submodule` to point at a newer version...  and upstream already includes a few other changes AFAICT. 

Ewen

mithro, 2018-01-18:
@ewen-naos-nz Merged the enjoy-digital/litex pull request -- https://github.com/enjoy-digital/litex/pull/52

ewen-naos-nz, 2018-01-18:
@mithro [timvideos/litex-buildenv#10](https://github.com/timvideos/litex-buildenv/pull/10) for our side of the change.  At least with `atftpd` this allows running without root.

It might still be worth providing, eg, Python-based TFTP server to reduce need to install system dependencies that auto-start daemons.  (But I suspect some users will be forced to interact with their firewall config anyway.)

ewen-naos-nz, 2018-01-18:
Of note, the "built in" qemu TFTP server (used in `scripts/build-qemu.sh` in the `user` network config) currently only runs on UDP/69, so is relying on BIOS fallback to trying UDP/69.  If someone can figure out a way to tell it to listen on a different port, it should now have `TFTP_SERVER_PORT` in its environment to know which port is intended.

mithro, 2018-01-18:
This has been mostly done thanks to @ewen-naos-nz -- I'll create a seperate issue for the Python-based TFTP server.

closed 2018-01-18 by mithro

ewen-naos-nz, 2018-01-18:
tftpy code appears to be at https://github.com/msoulier/tftpy now.  Last update 2017 (cf sf.net page last updated around 2010).  Seems to have a `listenport` option on the TFTPServer class, so at least could be given a simple frontend to listen on a different port.

py3tftp (https://github.com/sirMackk/py3tftp)  is server only, takes port on command line, and defaults to UDP/9069.

Potentially either one would do as a "simple TFTP Server run as user".