timvideos/litex-buildenv #14: third_party/qemu-litex not automatically updated if changed

ewen-naos-nz, 2018-01-20:
Sometimes `third_party/qemu_litex` needs updates to match with changes to other litex components.  Eg, 2018-01-20 ethernet updates from `third_party/qemu-litex` in order to successfully TFTP boot arty, otherwise boot hung around the point it started trying to send network packets.

However (a) `third_party/qemu-litex` is not a git submodule, just a git checkout in `scripts/build-qemu.sh` and (b) `scripts/build-qemu.sh` does not automatically fetch newer updates to `third_part/qemu-litex` (or up to a specific "known good" commit, as happens with `git submodule` references when updating), which means it will not automatically be updated.  (Instead the user will need to do `( cd third_party/qemu-litex && git pull)` manually, and find out that's needed some other way.)

ewen-naos-nz, 2018-01-20:
For benefit of FPGA Miniconf users: there is such a "required commit" to `third_party/qemu-litex` to match changes in litex merged around 2018-01-19/2018-01-20.  The required commit is:

    commit dc90b3543ce0eb06814b7bccb391478621bf698f
    Author: Tim 'mithro' Ansell <me@mith.ro>
    Date:   Thu Jan 4 00:08:31 2018 +0000

        liteeth: Fix for changes in LiteEth upstream.

which wasn't merged until more recently than that, so depending on when you started preparation for FPGA Miniconf you may have changes in some parts of `litex-buildenv`, but not that change.

Ewen

mithro, 2018-01-20:
Yeah - your a bit damned if you do, damned if you don't with submodules stuff.

Once https://github.com/timvideos/litex-buildenv/pull/15 is merged the `scripts/build-qemu.sh` will pull the latest changes in the same way the `scripts/build-linux.sh` script does.

PR #15 (2018-01-21): Random collection of fixes.
Bunch of fixes both extracted from @jimmo temporary commits and added while getting LiteEth to boot in QEMU for @shenki.

Fixes #14.
closed 2018-01-21 by mithro