timvideos/litex-buildenv #23: lm32 mimasv2 is broken

futaris, 2018-02-10:
Fresh checkout

revision https://github.com/timvideos/litex-buildenv/commit/ddd71f41b1e4942610de5e3f6cac667a6d2dd53b doesn't work
revision https://github.com/timvideos/litex-buildenv/commit/d11f5b85e7a962b3b3e55aaee70a693159f59957 doesn't work

Gets stuck in the BIOS:
`LiteX SoC BIOS (`

ewen-naos-nz, 2018-02-10:
I've just rebuilt gateware and micropython for Mimas v2 on [`786ee03fbf59465e6610a1a8b687bf2677292269`](https://github.com/timvideos/litex-buildenv/commit/786ee03fbf59465e6610a1a8b687bf2677292269) which I happened to have checked out, and after "make gateware-flash" and "make firmware-load" found that I got no response at all on serial port to reset. 

If I try "make firmware-load" with that new build on my other Mimas v2 board, which has older gateware already loaded, then that works.  So it does feel like Mimas v2 gateware might be broken in some way at some point recently.

The last commit I can be certain I did a full rebuild/install for Mimas v2 was [`3a162bb88e0717155ee9deedc23b4d6c156d6d70`](https://github.com/timvideos/litex-buildenv/commit/3a162bb88e0717155ee9deedc23b4d6c156d6d70), which was when I wrote the blog post about it.  Which is about 3 weeks ago.

I'll do some rebuilding and see if I can figure out where we broke it...

futaris, 2018-02-10:
https://github.com/timvideos/litex-buildenv/commit/d11f5b85e7a962b3b3e55aaee70a693159f59957 works with "make image-flash" . Don't do "make gateware-flash" and "make firmware-load" ...

https://logs.timvideos.us/%23timvideos/%23timvideos.2018-02-10.log.html

ewen-naos-nz is going to update the instructions slightly at:
http://ewen.mcneill.gen.nz/blog/entry/2018-01-17-fupy-fpga-micropython-on-mimas-v2-and-arty-a7/

ewen-naos-nz, 2018-02-10:
Based on some interactive debbuging on IRC (`#timvideos`) we've concluded that the instructions in [my blog post about MicroPython](http://ewen.mcneill.gen.nz/blog/entry/2018-01-17-fupy-fpga-micropython-on-mimas-v2-and-arty-a7/) only worked on the Mimas v2 if you happen to have installed something on the Mimas v2 before that already included the BIOS/Firmware.  Otherwise you end up with gateware and no BIOS/Firmware (or incomplete BIOS/Firmware), and there's nothing for "`make firmware-load`" to talk to.  Presumably the board I first tested it on happened to already have had "`make image-flash`" done.

I'm updating the blog post now... :-)

Ewen

closed 2018-02-10 by futaris

ewen-naos-nz, 2018-02-10:
Blog post updated.

I'm leaning towards thinking that we want a Makefile target that installs Gateware + BIOS but no application to reduce flashing times on Mimas v2 at 19.2kbps, leaving application to be serial booted only.  As I mentioned on IRC I think [I did that last year, but perhaps only by hand](http://ewen.mcneill.gen.nz/blog/entry/2017-03-12-micropython-on-numato-mimas-v2/).

Ewen

ewen-naos-nz, 2018-02-10:
FTR, now that https://github.com/timvideos/litex-buildenv/pull/24 is merged, in theory "`make gateware-flash`"/"`make firmware-load`" on Mimas v2 will work the way I kept believing they did :-)   ("`gateware-flash`" on Mimas v2 will now flash `image+gateware+bios+none.bin`.)

Ewen