timvideos/litex-buildenv #42: Issue in nexys_video gateware with RS232PHYTX

mithro, 2018-08-28:
```
./build/nexys_video_net_lm32/output.20180828-035725.log
---------------------------------------------
Traceback (most recent call last):
  File "./make.py", line 143, in <module>
    main()
  File "./make.py", line 108, in main
    soc = SoC(platform, ident=SoC.__name__, **soc_sdram_argdict(args), **dict(args.target_option))
  File "/home/travis/build/timvideos/HDMI2USB-litex-firmware/targets/nexys_video/net.py", line 29, in __init__
    BaseSoC.__init__(self, platform, *args, **kwargs)
  File "/home/travis/build/timvideos/HDMI2USB-litex-firmware/targets/nexys_video/base.py", line 121, in __init__
    self.submodules.uart_multiplexer = UARTMultiplexer(uart_interfaces, self.uart_phy)
  File "/home/travis/build/timvideos/HDMI2USB-litex-firmware/third_party/litex/litex/soc/cores/uart.py", line 245, in __init__
    uart.tx.eq(uarts[n].tx),
  File "/home/travis/build/timvideos/HDMI2USB-litex-firmware/third_party/migen/migen/fhdl/module.py", line 136, in __getattr__
    raise AttributeError("'"+self.__class__.__name__+"' object has no attribute '"+name+"'")
AttributeError: 'RS232PHYTX' object has no attribute 'eq'
```

enjoy-digital, 2018-08-28:
You need to use:

``self.submodules.uart_multiplexer = RS232PHYMultiplexer(uart_interfaces, self.uart_phy)``

enjoy-digital, 2018-08-28:
@mithro: can you give me permission to the repo so that i can do these changes directly when i'm refactoring code?

mithro, 2018-08-29:
@enjoy-digital I believe this is done now. Can you check?

closed 2018-08-29 by mithro