timvideos/litex-buildenv #60: Issue with updated migen+litedram

mithro, 2018-09-17:
```
Traceback (most recent call last):
  File "./make.py", line 164, in <module>
    main()
  File "./make.py", line 123, in main
    soc = get_soc(args, platform)
  File "./make.py", line 57, in get_soc
    soc = SoC(platform, ident=SoC.__name__, **soc_sdram_argdict(args), **dict(args.target_option))
  File "/home/travis/build/timvideos/litex-buildenv/targets/atlys/net.py", line 32, in __init__
    BaseSoC.__init__(self, platform, *args, **kwargs)
  File "/home/travis/build/timvideos/litex-buildenv/targets/atlys/base.py", line 228, in __init__
    dqs_ddr_alignment="C0")
  File "/home/travis/build/timvideos/litex-buildenv/third_party/litedram/litedram/phy/s6ddrphy.py", line 67, in __init__
    self.dfi = Interface(addressbits, bankbits, 2*databits, nphases)
  File "/home/travis/build/timvideos/litex-buildenv/third_party/litedram/litedram/phy/dfi.py", line 45, in __init__
    Record.__init__(self, layout)
  File "/home/travis/build/timvideos/litex-buildenv/third_party/migen/migen/genlib/record.py", line 106, in __init__
    finst = Record(fsublayout, prefix + fname, **kwargs)
  File "/home/travis/build/timvideos/litex-buildenv/third_party/migen/migen/genlib/record.py", line 103, in __init__
    finst = Signal(fsize, name=prefix + fname, **kwargs)
  File "/home/travis/build/timvideos/litex-buildenv/third_party/migen/migen/fhdl/structure.py", line 384, in __init__
    reset = Constant(reset, (self.nbits, self.signed))
  File "/home/travis/build/timvideos/litex-buildenv/third_party/migen/migen/fhdl/structure.py", line 295, in __init__
    raise TypeError("Width must be a strictly positive integer")
TypeError: Width must be a strictly positive integer
```

mithro, 2018-09-17:
@enjoy-digital

enjoy-digital, 2018-09-17:
Sorry, this should fix it: https://github.com/enjoy-digital/litedram/commit/d23dbf6e57a5037cc52beb16a6b58c8e55f7298e.

mithro, 2018-09-30:
Fixed.

closed 2018-09-30 by mithro