timvideos/litex-buildenv #61: Issue with updated liteeth - `NameError: name 'inc_cases' is not defined`

mithro, 2018-09-17:
```
Traceback (most recent call last):
  File "./make.py", line 164, in <module>
    main()
  File "./make.py", line 123, in main
    soc = get_soc(args, platform)
  File "./make.py", line 57, in get_soc
    soc = SoC(platform, ident=SoC.__name__, **soc_sdram_argdict(args), **dict(args.target_option))
  File "/home/travis/build/timvideos/litex-buildenv/targets/arty/net.py", line 39, in __init__
    phy=self.ethphy, dw=32, interface="wishbone")
  File "/home/travis/build/timvideos/litex-buildenv/third_party/liteeth/liteeth/core/mac/__init__.py", line 30, in __init__
    self.submodules.interface = LiteEthMACWishboneInterface(dw, nrxslots, ntxslots, endianness)
  File "/home/travis/build/timvideos/litex-buildenv/third_party/liteeth/liteeth/core/mac/wishbone.py", line 19, in __init__
    self.submodules.sram = sram.LiteEthMACSRAM(dw, sram_depth, nrxslots, ntxslots, endianness)
  File "/home/travis/build/timvideos/litex-buildenv/third_party/liteeth/liteeth/core/mac/sram.py", line 268, in __init__
    self.submodules.writer = LiteEthMACSRAMWriter(dw, depth, nrxslots, endianness)
  File "/home/travis/build/timvideos/litex-buildenv/third_party/liteeth/liteeth/core/mac/sram.py", line 31, in __init__
    inc_cases["default"] = inc.eq(4)
NameError: name 'inc_cases' is not defined
```

mithro, 2018-09-17:
@enjoy-digital

enjoy-digital, 2018-09-17:
Sorry, this should fix it: https://github.com/enjoy-digital/liteeth/commit/3d868449e9c38a00524cff8ed2bf5dec2fc0d858.

mithro, 2018-09-30:
Fixed.

closed 2018-09-30 by mithro