timvideos/litex-buildenv #68: Make targets auto-generated?

mithro, 2018-09-23:
There is a huge amount of duplication between the various targets.

For example the [base.py of the Atlys](https://github.com/timvideos/litex-buildenv/blob/master/targets/atlys/base.py) is almost identical to the [Opsis base.py](https://github.com/timvideos/litex-buildenv/blob/master/targets/opsis/base.py).

As we are about to add a large number of iCE40 devices in #29 - the amount of duplication is going to increase a lot.

@enjoy-digital thought it could be auto-generated pretty easily.

mithro, 2018-09-23:
Basically the code could be something like;
```python
if platform.hasattr("feature exists"):
    platform.add("feature")
```

mithro, 2018-09-23:
Some things which are different between boards are;
 * Types of DDR models
 * Types of SPI models
 * Clock and reset generation

cr1901, 2018-09-23:
This is probably a complement issue to #28.