timvideos/litex-buildenv #70: Install problems ARCH Linux X64

diginfo, 2018-09-26:
Just received the Arty A7: Artix-7  and was keen to get started but failed at the first hurdle !

After cloning, make install fails with:

`make: *** No rule to make target 'install'.  Stop.`

Followed by:

```
# make test   
python3 -m "hdmi2usb.modeswitch.tests"
WARNING:root:unbind-helper not found, will have to run as root!
/dev/bus/usb/001/001 -- lib: LibDevice(1d6b:0002:0414 '0000:02:01.0')                     ls: LsusbDevice(1d6b:0002:0414 '0000:02:01.0')                   -- /dev/bus/usb/001/001                      drivers: ('usb',)
/dev/bus/usb/002/001 -- lib: LibDevice(1d6b:0001:0414 '0000:02:00.0')                     ls: LsusbDevice(1d6b:0001:0414 '0000:02:00.0')                   -- /dev/bus/usb/002/001                      drivers: ('usb',)
/dev/bus/usb/002/002 -- lib: LibDevice(0e0f:0003:0103 /dev/bus/usb/002/002)               ls: LsusbDevice(0e0f:0003:0103 /dev/bus/usb/002/002)             -- /dev/bus/usb/002/002                      drivers: ('usbhid',)
/dev/bus/usb/002/003 -- lib: LibDevice(0e0f:0002:0100 /dev/bus/usb/002/003)               ls: LsusbDevice(0e0f:0002:0100 /dev/bus/usb/002/003)             -- /dev/bus/usb/002/003                      drivers: ('hub',)
python3 hdmi2usb/modeswitch/files.py hdmi2usb/firmware/spartan6/atlys/bscan_spi_xc6slx45.bit
XilinxBitFile(ncdname='bscan_spi_xc6slx45.ncd;UserID=0xFFFFFFFF', part='6slx45csg324', date='2015/07/19 15:57:31')
python3 setup.py test
/nfs/qnap/data/uPython/fupy/HDMI2USB-mode-switch/conda/lib/python3.7/site-packages/setuptools/dist.py:398: UserWarning: Normalizing 'v0.0.1.post31' to '0.0.1.post31'
  normalized_version,
running test
running egg_info
writing hdmi2usb.modeswitch.egg-info/PKG-INFO
writing dependency_links to hdmi2usb.modeswitch.egg-info/dependency_links.txt
writing entry points to hdmi2usb.modeswitch.egg-info/entry_points.txt
writing top-level names to hdmi2usb.modeswitch.egg-info/top_level.txt
reading manifest file 'hdmi2usb.modeswitch.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching 'hdmi2usb/_version.py'
writing manifest file 'hdmi2usb.modeswitch.egg-info/SOURCES.txt'
running build_ext
WARNING:root:unbind-helper not found, will have to run as root!
/dev/bus/usb/001/001 -- lib: LibDevice(1d6b:0002:0414 '0000:02:01.0')                     ls: LsusbDevice(1d6b:0002:0414 '0000:02:01.0')                   -- /dev/bus/usb/001/001                      drivers: ('usb',)
/dev/bus/usb/002/001 -- lib: LibDevice(1d6b:0001:0414 '0000:02:00.0')                     ls: LsusbDevice(1d6b:0001:0414 '0000:02:00.0')                   -- /dev/bus/usb/002/001                      drivers: ('usb',)
/dev/bus/usb/002/002 -- lib: LibDevice(0e0f:0003:0103 /dev/bus/usb/002/002)               ls: LsusbDevice(0e0f:0003:0103 /dev/bus/usb/002/002)             -- /dev/bus/usb/002/002                      drivers: ('usbhid',)
/dev/bus/usb/002/003 -- lib: LibDevice(0e0f:0002:0100 /dev/bus/usb/002/003)               ls: LsusbDevice(0e0f:0002:0100 /dev/bus/usb/002/003)             -- /dev/bus/usb/002/003                      drivers: ('hub',)

----------------------------------------------------------------------
Ran 0 tests in 0.000s

OK
make -C udev test
make[1]: Entering directory '/nfs/qnap/data/uPython/fupy/HDMI2USB-mode-switch/udev'
Checking hdmi2usb-human-path-helper.sh..
/bin/sh: line 2: posh: command not found
make[1]: *** [Makefile:44: test] Error 1
make[1]: Leaving directory '/nfs/qnap/data/uPython/fupy/HDMI2USB-mode-switch/udev'
make: *** [Makefile:64: test] Error 2
```
So it looks like this needs powershell, which I attempted to install but the install got trapped in a endless loop with 100% cpu usage for around an hour before I killed it.

As this is only to create som udev rules, can I create these rules manually ?

Thanks

mithro, 2018-09-26:
What instructions are you following?

There are two guides for the Arty A7 on the wiki;
https://github.com/timvideos/litex-buildenv/wiki

diginfo, 2018-09-26:
I am following these instructions:

https://github.com/timvideos/litex-buildenv/wiki/HowTo-FuPy-on-a-Digilent-Arty-A7

Just received the arty-a7 and are trying to run fupy on arch linux 64

All of my other development is done on arch 64 (esp32, pyboard, esp8266, lobo/esp32) and I would prefer not to have to create a ubuntu VM just to run fupy :-)

CarlFK, 2018-09-26:
> GitHub
> *timvideos/litex-buildenv <https://github.com/timvideos/litex-buildenv>*
> An environment for building LiteX based FPGA designs. Makes it easy to get
> everything you need! - timvideos/litex-buildenv

> As this is only to create som udev rules, can I create these rules
manually ?

yes, but it seems troubling that it gets stuck.  we would much prefer
figuring out what is broken.

> ...powershell, which I attempted to install but the install got trapped
in a endless loop

what?

ps - I am barely qualified to help, I mostly try to clean up the docs and
keep track of what is current vs deprecated.

diginfo, 2018-09-26:
```
Checking hdmi2usb-human-path-helper.sh..
/bin/sh: line 2: posh: command not found
make[1]: *** [Makefile:44: test] Error 1
make[1]: Leaving directory '/nfs/qnap/data/uPython/fupy/HDMI2USB-mode-switch/udev'
make: *** [Makefile:64: test] Error 2
```

> /bin/sh: line 2: posh: command not found

As you can see the script failed trying to execute posh, which I assume is a dependancy for powershell ?

diginfo, 2018-09-27:
If I just run make, then the building of the scripts fails with "W504 line break after binary operator" errors:

```
hdmi2usb/modeswitch/boards.py:237:13: W504 line break after binary operator
hdmi2usb/modeswitch/cli.py:246:21: W504 line break after binary operator
hdmi2usb/modeswitch/cli.py:297:31: W504 line break after binary operator
hdmi2usb/modeswitch/cli.py:298:31: W504 line break after binary operator
hdmi2usb/modeswitch/cli.py:299:31: W504 line break after binary operator
hdmi2usb/modeswitch/cli.py:300:31: W504 line break after binary operator
hdmi2usb/modeswitch/cli.py:301:31: W504 line break after binary operator
hdmi2usb/modeswitch/cli.py:302:31: W504 line break after binary operator
hdmi2usb/modeswitch/cli.py:371:17: W504 line break after binary operator
hdmi2usb/modeswitch/cli.py:372:17: W504 line break after binary operator
hdmi2usb/modeswitch/cli.py:373:17: W504 line break after binary operator
```