timvideos/litex-buildenv #71: Unable to locate Vivado directory or settings.

diginfo, 2018-09-27:
I find this very confusing, is it necessary to install Xilinx Vivado WebPACK FPGA synthesis tool ??

Further down the paragraph it states "We will not install Vivado for this guide"

> To customise your own gateware you will need the Xilinx Vivado WebPACK FPGA synthesis tool installed and reachable from /opt/Xilinx. The "Webpack" version of Xilinx Vivado, which can be run with a no-cost license file, is sufficient for litex-buildenv. We will not install Vivado for this guide. Instead, let's download a prebuilt gateware image:

when I run "make gateware-load" it allways fails with:

```
    _run_vivado(build_name, toolchain_path, source)
  File "/root/litex-buildenv/third_party/litex/litex/build/xilinx/vivado.py", line 78, in _run_vivado
    raise OSError("Unable to locate Vivado directory or settings.")
OSError: Unable to locate Vivado directory or settings.

real	0m2.180s
user	0m1.676s
sys	0m0.120s
Makefile:256: recipe for target 'gateware' failed
make[1]: *** [gateware] Error 1
make[1]: Leaving directory '/root/litex-buildenv'
Makefile:269: recipe for target 'build/arty_base_lm32//gateware/top.bit' failed
make: *** [build/arty_base_lm32//gateware/top.bit] Error 2
```

And it appears that it cannot locate the Vivado directory hence my initial question ?!

mithro, 2018-09-27:
@diginfo - if you are getting this message then it means the build system is trying to build the gateware (so it can be loaded).

This likely means something has gone wrong with the  earlier downloading pre-built gateware steps. Could you please try those a again and post any error you get for that.

diginfo, 2018-09-27:
Thanks, did not notice any errors previously, but will try again shortly.

But can you confirm whether or not I have to install the 17GB behemoth of vivado which means my hard disk is too small !!

diginfo, 2018-09-27:
Did not notice this failing before, but it does now !

```
# ./scripts/download-prebuilt.sh
Running the script as root is not permitted
Build directory 'build/arty_base_lm32/' already exists.
Remove this directory with 'make clean' before running this script.
(LX P=arty T=base F=micropython) root@ubuntu:~/litex-buildenv# make clean
rm -f build/cache.mk
rm -rf build/arty_base_lm32/
py3clean . 2>/dev/null || rm -rf $(find -name __pycache__)
(LX P=arty T=base F=micropython) root@ubuntu:~/litex-buildenv# 
(LX P=arty T=base F=micropython) root@ubuntu:~/litex-buildenv# 
(LX P=arty T=base F=micropython) root@ubuntu:~/litex-buildenv# ./scripts/download-prebuilt.sh
Running the script as root is not permitted

   Prebuilt Repository: timvideos/HDMI2USB-firmware-prebuilt

       Upstream remote: origin
       Upstream branch: master
      Upstream git rev: 1a111fa023c97df7c9c88a60b057477a57bd63a0
 Upstream git describe: v0.0.4-324-g1a111fa

              Platform: arty
                Target: base (default: net)
                   CPU: lm32 (default: lm32)
               Firmare: micropython (default: firmware)

Downloading
 from 'https://github.com/timvideos/HDMI2USB-firmware-prebuilt//trunk/archive/master/v0.0.4-324-g1a111fa/arty/base/lm32/'
   to 'build/arty_base_lm32/'
---------------------------------------
svn: E170000: URL 'https://github.com/timvideos/HDMI2USB-firmware-prebuilt/trunk/archive/master/v0.0.4-324-g1a111fa/arty/base/lm32' doesn't exist
```

mithro, 2018-09-27:
If you wish to modify the gateware (IE the virtual hardware inside the FPGA) then you will need to install Vivado as it is currently required for that.

If you are happy to just work on the firmware that runs on the gateware, then it won't be needed.

diginfo, 2018-09-27:
https://github.com/timvideos/HDMI2USB-firmware-prebuilt/tree/master/archive/master/v0.0.4-322-g3548304

seems to be the last archive, however v0.0.4-324-g1a111fa does not appear to exist !

diginfo, 2018-09-27:
if you can let me know the correct path I will temporarily hard-code it into ./scripts/download-prebuilt.sh

mithro, 2018-09-27:
It looks like this version has not built yet. `v0.0.4-322-g3548304` looks like the latest version.

mithro, 2018-09-27:
That looks like an issue in the download prebuilt script. The `/home/travis/...` path should have been translated to your local path.

diginfo, 2018-09-27:
OK, started again and have successfully completed all steps and are able to get a REPL.

But I have a few questions:

1) I tried to import machine but that failed, what I/O etc are available on the device ?
2) Now that upy has been uploaded, can I get a REPL using a serial terminal and when I quit out of the REPL how do I get back in again ?

If all of this is documented somewhere maybe you can point me to it :-)

mithro, 2018-09-27:
The pull request at https://github.com/timvideos/litex-buildenv/pull/72 should fix the issue with the absolute paths. It will be merged shortly.

mithro, 2018-09-27:
@diginfo The best place to ask the questions about actually using MicroPython is on the [#timvideos IRC channel](http://webchat.freenode.net?channels=%23timvideos).

The MicroPython port is very much still a work in progress and we would love help improving it!