timvideos/litex-buildenv #100: icebreaker doesn't set port so make firmware-connect fails

mithro, 2018-11-13:
```
(LX P=icebreaker.minimal F=stub R=master-lbe) tansell@tansell:~/github/timvideos/litex-buildenv$ make firmware-connect
flterm --port= --speed=115200
[FLTERM] Starting...
Unable to open serial port: No such file or directory
```

ewenmcneill, 2019-02-22:
Weirdly, I was getting that kind of message on MimasV2 and Arty just now, while trying to test something else; I'm a bit unclear what's changed in the way "make firmware-connect" is handled...

Just mentioning it in case this is not Icebreaker specific.

Ewen

ewenmcneill, 2019-02-25:
In https://github.com/timvideos/litex-buildenv/issues/100#issuecomment-466281982 I wrote:
> Weirdly, I was getting that kind of message on MimasV2 and Arty just now, while trying to test something else; I'm a bit unclear what's changed in the way "make firmware-connect" is handled...

I figured out how this was happening.  My wrapper script for entering environments was unconditionally setting `COMM_PORT`to the value passed in (for TinyFPGA BX support, where it's not pre-defined), even when no value was passed in.  But the `Makefile`s only set `COMM_PORT` if it is *unset* (not if it is empty).  So it was not getting set when I ended up with an explicitly empty `COMM_PORT` value in the environment.  When I fix this bit of my wrapper script to conditionally pass in `COMM_PORT` only if defined, my Mimas v2 (and I'd guess Arty) cases work again for me.

I suspect the Icebreaker and TinyFPGA BX (both of which do not set `COMM_PORT`) probably need something to catch it being unset and show a more useful error message (eg, `export COMM_PORT=....`).

Ewen