timvideos/litex-buildenv #99: iceprog isn't provided in conda so icebreaker fails

mithro, 2018-11-13:
icebreaker programming fails because of missing iceprog.

It should be part of conda.

cr1901, 2018-11-16:
https://github.com/timvideos/conda-hdmi2usb-packages/blob/master/icestorm/noiceprog.patch Getting rid of this patch should fix the issue. Do you recall why you had it in the first place?

As for why it "worked on my machine":

```
(LX P=tinyfpga_bx.minimal F=micropython) william@xubuntu-dtrain:~/Projects/litex-buildenv$ iceprog
iceprog: missing argument
Try `iceprog --help' for more information.
(LX P=tinyfpga_bx.minimal F=micropython) william@xubuntu-dtrain:~/Projects/litex-buildenv$ which iceprog
/home/william/.local/bin/iceprog
(LX P=tinyfpga_bx.minimal F=micropython) william@xubuntu-dtrain:~/Projects/litex-buildenv$ which icebram
/home/william/Projects/litex-buildenv/build/conda/bin/icebram
(LX P=tinyfpga_bx.minimal F=micropython) william@xubuntu-dtrain:~/Projects/litex-buildenv$ 
```

Looks like the build environment is falling back to my locally-installed version.

cr1901, 2018-11-16:
@mithro I'm pretty sure we just assume the host provides `libftdi`/`libusb`<sup>1</sup>? In any case both `libftdi`/`libusb` are certainly already required to program arty boards. So I don't see the harm in bringing in `iceprog`.

1.  For now anyway. In my dormant conda-on-windows branch, I ended up building/packaging `libftdi`/`libusb` for Windows users.

mithro, 2018-11-16:
Yes - `libftdi` / `libusb` shouldn't be required for icestorm. They can be required for iceprog.

ewenmcneill, 2019-07-25:
After doing `conda update --all` (which downgraded some things) to get a consistent environment after `download-env.sh` and `conda install binutils-riscv32-elf` had done their thing, it's now failing with a different dependency.

My guess is that the *newer* uploads of `iceprog` aren't even being considered due to other dependency problems, and the older uploads of `iceprog` will no longer install because they were built for Python 3.7 (.0?) which is no longer available in Conda.  And/or because it's picking at random -- I can't pin a specific version to install, AFAICT, because they're all the same (upstream) version, just built at different times.

Ewen

```
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/src/fpga/litex-buildenv$ conda install iceprog
Collecting package metadata (current_repodata.json): done
Solving environment: failed with current_repodata.json, will retry with next repodata source.
Initial quick solve with frozen env failed.  Unfreezing env and trying again.
Solving environment: failed with current_repodata.json, will retry with next repodata source.

ResolvePackageNotFound: 
  - python==3.7

(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/src/fpga/litex-buildenv$ conda list | grep python
ipython                   7.6.1            py37h39e3cac_0  
ipython_genutils          0.2.0                    py37_0  
python                    3.7.3                h0371630_0  
python-libarchive-c       2.8                     py37_11  
python-utils              2.3.0                    pypi_0    pypi
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/src/fpga/litex-buildenv$
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/src/fpga/litex-buildenv$ conda info iceprog | egrep -A 2 "version" | grep -A 2 v0.0_0659_g8f61acd
WARNING: 'conda info package_name' is deprecated.
          Use 'conda search package_name --info'.
version     : v0.0_0659_g8f61acd
build string: 20190508_221239
build number: 20190508221239
--
version     : v0.0_0659_g8f61acd
build string: 20190504_161016
build number: 20190504161016
--
version     : v0.0_0659_g8f61acd
build string: 20190508_190655
build number: 20190508190655
--
version     : v0.0_0659_g8f61acd
build string: 20190504_160319
build number: 20190504160319
--
version     : v0.0_0659_g8f61acd
build string: 20190504_173522
build number: 20190504173522
--
version     : v0.0_0659_g8f61acd
build string: 20190504_170606
build number: 20190504170606
--
version     : v0.0_0659_g8f61acd
build string: 20190504_231101
build number: 20190504231101
--
version     : v0.0_0659_g8f61acd
build string: 20190509_015653
build number: 20190509015653
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/src/fpga/litex-buildenv$
```

ewenmcneill, 2019-07-25:
FTR, I eventually got it installed by manually downloading the latest conda `iceprog` tar file, and offline installing that:

```
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/tmp$ wget https://anaconda.org/TimVideos/iceprog/v0.0_0691_gdd8c003/download/linux-64/iceprog-v0.0_0691_gdd8c003-20190722_202536.tar.bz2
[...]
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/tmp$ conda install --offline /tmp/iceprog-v0.0_0691_gdd8c003-20190722_202536.tar.bz2 

Downloading and Extracting Packages
######################################################################## | 100% 
Preparing transaction: done
Verifying transaction: done
Executing transaction: done
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/tmp$ conda list | grep iceprog
iceprog                   v0.0_0691_gdd8c003 20190722_202536    <unknown>
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/tmp$ which iceprog
/src/fpga/litex-buildenv/build/conda/bin/iceprog
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/tmp$
```

and it seems to work to program an iCEBreaker.

Ewen

mithro, 2019-07-25:
@ewenmcneill - I kind of solved this a different way -- make no platform the default! :-P

ewenmcneill, 2019-07-25:
> @ewenmcneill - I kind of solved this a different way -- make no platform the default! :-P

@mithro, should we close the pull request that was trying to make TinyFPGA BX the default then?

IIRC there's a couple of other minor tweaks (and I have another one on my dev laptop to warn about `COMM_PORT` not being set), but they could be a different pull request.

Ewen

ewenmcneill, 2019-07-25:
FTR, having done all of that to get `iceprog` installed, `scripts/enter-env.sh` is upset that I don't have Python 3.7 (and it can't be installed) -- although I do have Python 3.7.3 -- so it bails out.

To even get to a point where `scripts/enter-env.sh` wouldn't complain again,  needed the patch below to the versions.   And even then `scripts/download-env.sh` fails (with `conda` version conflicts), so it's probably not the ideal solution to force a recent `iceprog` package build to installl...

Ewen

PS: Feels like we're going to need to have another go at getting our conda package dependencies straightened out, and possibly updated to Python 3.7.3 / gcc 9.  AT a quick glance the `libstdcxx-ng` versions required are the current sticking point for updating.

```
ewen@parthenon:/src/fpga/litex-buildenv$ git diff | cat
diff --git a/scripts/settings.sh b/scripts/settings.sh
index 4a11ab0..8aef854 100644
--- a/scripts/settings.sh
+++ b/scripts/settings.sh
@@ -6,7 +6,7 @@ THIRD_DIR=$TOP_DIR/third_party
 CONDA_DIR=$BUILD_DIR/conda
 
 CONDA_VERSION=4.6.14
-PYTHON_VERSION=3.7
+PYTHON_VERSION=3.7.3
 
 # Python module versions
 HDMI2USB_MODESWITCH_VERSION=0.0.1
@@ -14,8 +14,8 @@ HEXFILE_VERSION=0.1
 CMAKE_VERSION=3.14.4
 
 # Conda package versions
-BINUTILS_VERSION=2.31.1
-GCC_VERSION=8.2.0
+BINUTILS_VERSION=2.32
+GCC_VERSION=9.1.0
 SDCC_VERSION=3.5.0
 OPENOCD_VERSION=0.10.0
 
ewen@parthenon:/src/fpga/litex-buildenv$ 
```

```
Package libstdcxx-ng conflicts for:
asn1crypto -> python[version='>=2.7,<2.8.0a0,>=3.5,<3.6.0a0,>=3.6,<3.7.0a0,>=3.7,<3.8.0a0'] -> sqlite[version='>=3.20.1,<4.0a0,>=3.23.1,<4.0a0,>=3.25.2,<4.0a0'] -> libedit[version='>=3.1.20170329,<3.2.0a0,>=3.1.20181209,<3.2.0a0'] -> ncurses[version='6.0.*,>=6.1,<7.0a0'] -> libstdcxx-ng[version='>=7.2.0,>=7.3.0']
pexpect -> ptyprocess[version='>=0.5'] -> python[version='>=2.7,<2.8.0a0,>=3.5,<3.6.0a0,>=3.6,<3.7.0a0,>=3.7,<3.8.0a0'] -> sqlite[version='>=3.20.1,<4.0a0,>=3.23.1,<4.0a0,>=3.25.2,<4.0a0'] -> libedit[version='>=3.1.20170329,<3.2.0a0,>=3.1.20181209,<3.2.0a0'] -> ncurses[version='6.0.*,>=6.1,<7.0a0'] -> libstdcxx-ng[version='>=7.2.0,>=7.3.0']
[...]
```

```
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/src/fpga/litex-buildenv$ conda list | grep libstdcxx-ng
libstdcxx-ng              9.1.0                hdf63c60_0  
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/src/fpga/litex-buildenv$ 
```

mithro, 2019-07-25:
Hrm... 
```bash
function check_version {
        TOOL=$1
        VERSION=$2
        if $TOOL --version 2>&1 | grep -q $VERSION > /dev/null; then
                echo "$TOOL found at $VERSION"
                return 0
        else
                $TOOL --version
                echo "$TOOL (version $VERSION) *NOT* found"
                echo "Please try running the $SETUP_DIR/download-env.sh script again."
                return 1
        fi
}
```

Python 3.7.3 should match `PYTHON_VERSION=3.7`?

mithro, 2019-07-25:
Oh!

mithro, 2019-07-25:
What version of conda are you running? `4.7.10`?

mithro, 2019-07-25:
I'm guessing that you are missing the conda pin which keeps things at conda `4.6.10` because it is only added on the install of conda.

Can you try the following patch?
```patch
index 66d6cff..b1b51ea 100755
--- a/scripts/download-env.sh
+++ b/scripts/download-env.sh
@@ -219,13 +219,13 @@ export PATH=$CONDA_DIR/bin:$PATH:/sbin
                        ./Miniconda3-latest-Linux-x86_64.sh -p $CONDA_DIR -b -f || exit 1
                )
                fix_conda
-               conda config --system --set always_yes yes
-               conda config --system --set changeps1 no
                conda config --system --add envs_dirs $CONDA_DIR/envs
                conda config --system --add pkgs_dirs $CONDA_DIR/pkgs
-               pin_conda_package conda ${CONDA_VERSION}
-               conda update -q conda
        fi
+       conda config --system --set always_yes yes
+       conda config --system --set changeps1 no
+       pin_conda_package conda ${CONDA_VERSION}
+       conda update -q conda
        fix_conda
        conda config --system --add channels timvideos
        conda info
```

ewenmcneill, 2019-07-25:
I concluded that my `conda` environment had been messed up irretrievably (I couldn't even `conda uninstall iceprog` let alone anything else useful), so I've moved `build/conda` aside, and run `scripts/download-env.sh` again.  That seems to have worked (other than having to also remove `build/Miniconda3-latest-Linux-x86_64.sh` because the `wget --continue` has managed to download the 67MiB version by appending to the end of the the earlier 55MiB version...)

The result of doing all of that seems to end up with the expected `conda` 4.6 version:

```
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/src/fpga/litex-buildenv$ conda --version
conda 4.6.14
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/src/fpga/litex-buildenv$ 
```

so my guess is that at some point I did a `conda update --all` in the previous conda environment, perhaps while testing our conda dependencies, and that got my conda environment into a state where `iceprog` wouldn't install via conda.

With a fresh `scripts/download-env.sh` it *is* installed via conda:

```
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/src/fpga/litex-buildenv$ conda list iceprog
# packages in environment at /src/fpga/litex-buildenv/build/conda:
#
# Name                    Version                   Build  Channel
iceprog                   v0.0_0691_gdd8c003 20190722_202536    timvideos
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/src/fpga/litex-buildenv$ 
```

@mithro Maybe we close this ticket now?  `iceprog` *is* provided in conda, and is even installable if you don't upgrade to too recent conda... :-)

(Upgrading to more recent conda / Python 3.7.3, etc, can be separate.)

Ewen

PS: FTR, `scripts/download-env.sh` is getting Python 3.7.0 not 3.7.3:

```
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/src/fpga/litex-buildenv$ conda list python
# packages in environment at /src/fpga/litex-buildenv/build/conda:
#
# Name                    Version                   Build  Channel
ipython                   7.6.1            py37h39e3cac_0  
ipython_genutils          0.2.0                    py37_0  
python                    3.7.0                h6e4f718_3  
python-utils              2.3.0                    pypi_0    pypi
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/src/fpga/litex-buildenv$
```

mithro, 2019-07-25:
Yes, lets close this issue. Related pull requests
 - https://github.com/timvideos/litex-buildenv/pull/177
 - #178

closed 2019-07-25 by mithro