timvideos/litex-buildenv #108: enter-env.py doesn't set PYTHONPATH

xobs, 2018-12-09:
The script `enter-env.py` appears to check for module imports by simply doing "import [module]".  It does this without first setting `PYTHONPATH`.

None of the other documentation mentions `PYTHONPATH`, and seems to imply that the user has already set this variable.  When running `enter-env.py` without first setting `PYTHONPATH`, the user is told to run `download-env.sh`, however this doesn't do anything with `third_party/`.

Therefore, the best place to set it appears to be in `enter-env.py`, or tell the user to first set the `PYTHONPATH` variable themselves.

mithro, 2018-12-09:
The person should need to set `PYTHONPATH`. The modules from third_party are installed into the conda environment as part of the `download-env.sh` script.

mithro, 2018-12-09:
What made you think you needed to set `PYHONPATH`?

mithro, 2018-12-10:
The tinyusb branch references stuff I have yet to push upstream. You will have to add my remote temporarily;

```
cd third_party/litex
git remote add mithro git+ssh://github.com/mithro/litex.git
```