timvideos/litex-buildenv #109: `make.py` ignores `CPU` environment variable, won't build

xobs, 2018-12-10:
The python script `make.py` refuses to run, stating that `args.cpu_type is not None`.  I have patched `make.py` to print out the contents of the `CPU` environment variable, which it appears to see.  However, this variable is never actually set as part of the `args` object:

```sh
(LX P=tinyfpga_bx T=usb.lite F=tinyusb R=tinyusb-work) xobs@Cuboid:~/Code/tomu/tinyusb$ python --version
Python 3.6.0 :: Continuum Analytics, Inc.
(LX P=tinyfpga_bx T=usb.lite F=tinyusb R=tinyusb-work) xobs@Cuboid:~/Code/tomu/tinyusb$ git diff
diff --git a/make.py b/make.py
index 63355c3..80d9b10 100755
--- a/make.py
+++ b/make.py
@@ -14,6 +14,7 @@ def get_args(parser, platform='opsis', target='hdmi2usb', firmware='firmware'):

     soc_sdram_args(parser)
     parser.set_defaults(cpu_type=os.environ.get('CPU', 'lm32'))
+    print("CPU env: " + os.environ.get('CPU', 'unset'))
     parser.set_defaults(cpu_variant=os.environ.get('CPU_VARIANT', None) or None)

     parser.add_argument("--iprange", default="192.168.100")
Submodule third_party/valentyusb contains untracked content
(LX P=tinyfpga_bx T=usb.lite F=tinyusb R=tinyusb-work) xobs@Cuboid:~/Code/tomu/tinyusb$ CPU=lm32 python ./make.py
CPU env: lm32
Traceback (most recent call last):
  File "./make.py", line 174, in <module>
    main()
  File "./make.py", line 136, in main
    builddir = get_builddir(args)
  File "./make.py", line 34, in get_builddir
    assert args.cpu_type is not None
AssertionError
(LX P=tinyfpga_bx T=usb.lite F=tinyusb R=tinyusb-work) xobs@Cuboid:~/Code/tomu/tinyusb$
```

mithro, 2018-12-10:
What command are you running?

xobs, 2018-12-10:
As mentioned on IRC, I was running `./make.py` because that's how upstream litex projects do it, such as genesys2-soc and nexys-soc, as well as most other litex-based projects such as netv2 and tinyfpga-soc.

With litex-buildenv, the preferred method seems to be to use make, but this isn't mentioned in the README anywhere.

mithro, 2018-12-10:
Yeah, looks like we lost the info around using `make` when we redid the `README.md` / `getting-started.md`

mithro, 2018-12-10:
Opened #110 and closing this one.

closed 2018-12-10 by mithro