timvideos/litex-buildenv #120: Compilation issue with Conda's lm32 toolchain

jeanthom, 2018-12-29:
Hi !

I'm running into weird issue with the lm32 toolchain that litex-buildenv uses. The compiled firmware isn't working correctly, the firmware hangs just before or midway through printf calls ; however the BIOS works OK!

[image]

@enjoy-digital sent me a toolchain ([http://www.das-labor.org/files/madex/lm32_linux_i386.tar.bz2](http://www.das-labor.org/files/madex/lm32_linux_i386.tar.bz2)) that works OK for me.

I am running litex-buildenv inside Xubuntu 18.04 virtualized using VirtualBox.

Attached to this issue are enjoy-digital/Chubby75 firmwares compiled with litex-buildenv and das-labor's toolchain :
[firmware-litex-buildenv.zip](https://github.com/timvideos/litex-buildenv/files/2716272/firmware-litex-buildenv.zip)
[firmware-das-labor.zip](https://github.com/timvideos/litex-buildenv/files/2716274/firmware-das-labor.zip)

jeanthom, 2019-05-27:
I can't reproduce the issue nor explain why it works for me now... I'm closing this issue.

closed 2019-05-27 by jeanthom

mithro, 2019-05-27:
@jeanthom I'm very interested in getting Chubby75 support in litex-buildenv repository. Are you interested in adding it?

jeanthom, 2019-05-28:
What should I do to provide Chubby75 support ? Adding platform & target scripts ?

mithro, 2019-05-28:
@jeanthom Yeap!