timvideos/litex-buildenv #125: make gateware fails on base target for platform ice40_up5k_b_evn

kamejoko80, 2019-04-02:
Hello,

I got FPGA placement error when building gateware for platform ice40_up5k_b_evn,  here are my steps below:

export CPU=vexriscv
export TARGET=base
export PLATFORM=ice40_up5k_b_evn
export FIRMWARE=micropython
source scripts/enter-env.sh
make gateware

Error log:

`Info: Device utilisation:
Info:            ICESTORM_LC:  8179/ 5280   154%
Info:           ICESTORM_RAM:    28/   30    93%
Info:                  SB_IO:    16/   96    16%
Info:                  SB_GB:     8/    8   100%
Info:           ICESTORM_PLL:     0/    1     0%
Info:            SB_WARMBOOT:     0/    1     0%
Info:           ICESTORM_DSP:     0/    8     0%
Info:         ICESTORM_HFOSC:     0/    1     0%
Info:         ICESTORM_LFOSC:     0/    1     0%
Info:                 SB_I2C:     0/    2     0%
Info:                 SB_SPI:     0/    2     0%
Info:                 IO_I3C:     0/    2     0%
Info:            SB_LEDDA_IP:     0/    1     0%
Info:            SB_RGBA_DRV:     0/    1     0%
Info:         ICESTORM_SPRAM:     4/    4   100%

Info: Placed 16 cells based on constraints.
Info: Creating initial placement for remaining 8219 cells.
Info:   initial placement placed 500/8219 cells
Info:   initial placement placed 1000/8219 cells
Info:   initial placement placed 1500/8219 cells
Info:   initial placement placed 2000/8219 cells
Info:   initial placement placed 2500/8219 cells
Info:   initial placement placed 3000/8219 cells
Info:   initial placement placed 3500/8219 cells
Info:   initial placement placed 4000/8219 cells
Info:   initial placement placed 4500/8219 cells
Info:   initial placement placed 5000/8219 cells
ERROR: failed to place cell '$abc$60802$auto$blifparse.cc:492:parse_blif$63173_LC' of type 'ICESTORM_LC'
ERROR: Placing design failed.
1 warning, 2 errors
Traceback (most recent call last):
  File "./make.py", line 164, in <module>
    main()
  File "./make.py", line 148, in main
    vns = builder.build(**dict(args.build_option))
  File "/home/phuong/Workspace/litex-buildenv/HDMI2USB-litex-firmware/third_party/litex/litex/soc/integration/builder.py", line 171, in build
    toolchain_path=toolchain_path, **kwargs)
  File "/home/phuong/Workspace/litex-buildenv/HDMI2USB-litex-firmware/third_party/litex/litex/soc/integration/soc_core.py", line 406, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/phuong/Workspace/litex-buildenv/HDMI2USB-litex-firmware/third_party/litex/litex/build/lattice/platform.py", line 29, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/phuong/Workspace/litex-buildenv/HDMI2USB-litex-firmware/third_party/litex/litex/build/lattice/icestorm.py", line 183, in build
    _run_script(script)
  File "/home/phuong/Workspace/litex-buildenv/HDMI2USB-litex-firmware/third_party/litex/litex/build/lattice/icestorm.py", line 64, in _run_script
    raise OSError("Subprocess failed")
OSError: Subprocess failed

real    2m26.098s
user    2m20.784s
sys     0m3.416s
Makefile:260: recipe for target 'gateware' failed
make: *** [gateware] Error 1
`

kamejoko80, 2019-04-02:
Oh, I see this line "Info: ICESTORM_LC: 8179/ 5280 154%", the ICESTORM_LC exceed maximum LC of the device. Is there any option to build vexriscv as minimul resource ?

mithro, 2019-04-02:
```bash
export CPU_VARIANT=minimal  # Use a resource-constrained variant to make up5k happy
```

mithro, 2019-04-02:
I guess we should also add a check which prevents non-`CPU_VARIANT=minimal` version on the iCE40?

kamejoko80, 2019-04-08:
@mithro 

Thanks for the reply
CPU_VARIANT=minimal seams don't work with vexriscv, however it is ok with lm32.
Below is the error build log:  

mkdir -p build/ice40_up5k_b_evn_base_vexriscv.minimal/
time python -u ./make.py --platform=ice40_up5k_b_evn --target=base --cpu-type=vexriscv --iprange=192.168.100    --cpu-variant=minimal  \
        2>&1 | tee -a /home/phuong/Workspace/litex-buildenv/HDMI2USB-litex-firmware/build/ice40_up5k_b_evn_base_vexriscv.minimal//output.20190408-224923.log; (exit ${PIPESTATUS[0]})
Traceback (most recent call last):
  File "./make.py", line 164, in <module>
    main()
  File "./make.py", line 123, in main
    soc = get_soc(args, platform)
  File "./make.py", line 57, in get_soc
    soc = SoC(platform, ident=SoC.__name__, **soc_sdram_argdict(args), **dict(args.target_option))
  File "/home/phuong/Workspace/litex-buildenv/HDMI2USB-litex-firmware/targets/ice40_up5k_b_evn/base.py", line 83, in __init__
    SoCCore.__init__(self, platform, clk_freq, **kwargs)
  File "/home/phuong/Workspace/litex-buildenv/HDMI2USB-litex-firmware/third_party/litex/litex/soc/integration/soc_core.py", line 187, in __init__
    self.add_cpu(vexriscv.VexRiscv(platform, self.cpu_reset_address, self.cpu_variant))
  File "/home/phuong/Workspace/litex-buildenv/HDMI2USB-litex-firmware/third_party/litex/litex/soc/cores/cpu/vexriscv/core.py", line 20, in __init__
    assert variant in variants, "Unsupported variant %s" % variant
AssertionError: Unsupported variant minimal

real    0m0.256s
user    0m0.240s
sys     0m0.012s
Makefile:260: recipe for target 'gateware' failed
make: *** [gateware] Error 1
(LX P=ice40_up5k_b_evn C=vexriscv.minimal F=micropython) phuong@ubuntu1604lts:~/Workspace/litex-buildenv/HDMI2USB-litex-firmware$

mithro, 2019-04-08:
[Looks like for vexriscv the valid values are;](https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/vexriscv/core.py#L19)
> "std", "std_debug", "lite", "lite_debug", "min", "min_debug"

Try export `CPU_VARIANT=min`

mithro, 2019-04-08:
Logged https://github.com/enjoy-digital/litex/issues/159 about having consistent variant naming across CPU types.

henrydang80, 2019-04-26:
It works, thanks

closed 2019-04-26 by mithro