timvideos/litex-buildenv #131: Bootstrap instructions are debian specific

mballance, 2019-05-03:
@mithro says "Everything else should work but the bootstrap is debian specific."

mithro, 2019-05-08:
Work in progress at https://github.com/timvideos/litex-buildenv/pull/132

mithro, 2019-05-10:
This has been fixed in #132. 

 * The `download-env-root.sh` script has been renamed to `debian-setup.sh`.
 * `bootstrap.sh` script only calls `debian-setup.sh` on Debian / Ubuntu systems. Otherwise it checks for the few things needed and tells you to install them.

TomKeddie, 2019-05-21:
@mithro  I think the grep -v is wrong, the -v at https://github.com/timvideos/litex-buildenv/blob/master/scripts/bootstrap.sh#L42 the logic I don't think we want this here.  On archlinux I see it try to run the debian script.
```
bash: line 43: ./scripts/debian-setup.sh: No such file or directory
```

Also when running bootstrap via curl as per [wiki](https://github.com/timvideos/litex-buildenv/wiki/Bootstrap) the other scripts won't be present.

mithro, 2019-05-21:
@TomKeddie You seem to be correct, it should be `grep -q` -- Do you want to submit your first pull request for fixing that?

mithro, 2019-05-21:
Regarding;
> Also when running bootstrap via curl as per [wiki](https://github.com/timvideos/litex-buildenv/wiki/Bootstrap) the other scripts won't be present.

Notice the script clones the github repository and then runs the scripts? See the following lines;

https://github.com/timvideos/litex-buildenv/blob/7fc049aad997d8deefb5980fc559fce5f2170675/scripts/bootstrap.sh#L33-L44

TomKeddie, 2019-05-21:
> Notice the script clones the github repository and then runs the scripts? See the following lines;

I see the problem now, the bootstrap instructions pull bootstrap.sh from litex-buildenv but it clones HDMI2USB-litex-firmware which doesn't have the changes, would you prefer the wiki point to HDMI2USB-litex-firmware for bootstrap?

mithro, 2019-05-21:
@TomKeddie I think LiteX-BuildEnv is probably the right default?

TomKeddie, 2019-05-21:
@mithro I'm not sure what the grand plan is but I think should should be consistent, should I do a PR for bootstrap.sh then?

mithro, 2019-05-21:
@TomKeddie We are making it up as we go along :-)

LiteX-BuildEnv is a rename of HDMI2USB-litex-firmware to indicate that it can be used for more than just the HDMI2USB-litex-firmware. At some point it would be nice to make HDMi2USB-litex-firmware just a derivative of the LiteX-BuildEnv which adds the HDMI2USB specific features but have not had the time to do that. (**Very** long term we would also like to probably rewrite a lot of LiteX-BuildEnv [as proposed in this document](https://docs.google.com/document/d/1V47Zc0eyjUUGBNsJuYORdb4TqH4C1WjzrX9odX0YdZE/edit#heading=h.50rcosdwubce)).

Sounds reasonable to do such a pull request.