timvideos/litex-buildenv #134: Change default board to iCE40 based board

ewenmcneill, 2019-05-11:
Now that we have a build chain that is fully open source (iCE40), we should change the default build target to a board that is iCE40 based.  Maybe TinyFPGA BX now, and iCEBreaker ones those are out/supported?

The reasoning here is that getting started with a Xilinx based board requires several non trivial manual steps (installing Vivado, or ISE), which are completely unnecessary for the iCE40 based boards, but users that just start following the defaults can quickly end up in the weeds doing a lot of yak shaving (eg, https://twitter.com/femtoduino/status/1126750242768969728).

By making the default (a) something it's now more likely they're trying to use and (b) something that gets to a "would work if you had that board" quickly, it's more likely they'll either get things working, or quickly figure out they're heading in the wrong direction.  (Currently it takes a lot of yak shaving to determine that installing lots of Xilinx things is the wrong direction.)

Capturing a suggestion that @mithro (https://twitter.com/mithro/status/1126988574547558400) and I (https://twitter.com/ewenmcneill/status/1126998432978165760) made independently on Twitter.

mithro, 2019-05-11:
SGTM!

ewenmcneill, 2019-07-25:
Closing issue, because the underlying surprise (that prompted this on IRC) got fixed by having *no* default platform -- see https://github.com/timvideos/litex-buildenv/pull/162.   More detail in https://github.com/timvideos/litex-buildenv/pull/138#issuecomment-514872522, https://github.com/timvideos/litex-buildenv/issues/99#issuecomment-514862433, https://github.com/timvideos/litex-buildenv/issues/99#issuecomment-514865663,  https://github.com/timvideos/litex-buildenv/issues/99#issuecomment-514882340, and https://github.com/timvideos/litex-buildenv/issues/99#issuecomment-514884125.

Ewen

closed 2019-07-25 by ewenmcneill