timvideos/litex-buildenv #157: Non-50% duty cycle output in mmcm.c output

mithro, 2019-07-01:
https://github.com/timvideos/litex-buildenv/blob/cd0ca838185c3aeac7f025cba8938ae33af64508/firmware/mmcm.c#L77-L78

```c
0x1000 | (10<<6) | 10
```
```c
0x1000 | (HIGH_AMOUNT<<6) | LOW_AMOUNT
```

(Should also make this into a macro rather...)

If `HIGH_AMOUNT` != `LOW_AMOUNT` (IE odd number) then the PLL's "EDGE bit" should be set so the output is still 50% duty cycle.

mithro, 2019-07-02:
Info here -> https://github.com/AlphamaxMedia/netv2-fpga/blob/0feddcb73281784f5b74b2fd6051860004c1c878/firmware/mmcm.c#L295-L302

rohitk-singh, 2019-07-04:
Yes, currently the duty cycle won't be true 50%. The edge bit needs to be set for true 50% duty cycle. This isn't a major issue with the logic which use only one clock-edge, but for ODDR kind of primitives, we better fix this.

[image]

We need to revamp the clock configuration API which can be used anywhere and not just with HDMI2USB. Can we design something we can be ported to multiple FPGAs too? Using MMCM structures was a step in right direction!