timvideos/litex-buildenv #172: on Windows wants udev rules

CarlFK, 2019-07-14:
last step of this section:

https://github.com/timvideos/litex-buildenv/wiki/Bash-On-Windows#install-litex

```
juser@DESKTOP-KOJ4LVF:~/lhub/litex-buildenv$ source ./scripts/enter-env.sh
Please install the HDMI2USB udev rules.
These are installed by scripts/download-env-root.sh
```
Does udev make sense on windows?

mithro, 2019-07-15:
udev doesn't make sense on Windows.

CarlFK, 2019-07-16:
this needs to ignore windows:

https://github.com/timvideos/litex-buildenv/blob/master/scripts/enter-env.sh#L63-L71

Later today I'll try to figure out how to see what OS we are running under

mithro, 2019-07-16:
@cr1901 - Any idea here?

cr1901, 2019-07-16:
```
William@William-THINK MINGW64 ~
$ uname -a
MINGW64_NT-6.1-7601 William-THINK 3.0.7-338.x86_64 2019-05-24 10:50 UTC x86_64 Msys
```

Prob best to check the `uname` string for a match for `NT` or something for now? Idk what the conda `bash` returns for `uname` on Windows...

CarlFK, 2019-07-20:
```
(LX P=arty C=mor1kx) juser@DESKTOP-KOJ4LVF:~/lhub/litex-buildenv$ uname -a
Linux DESKTOP-KOJ4LVF 4.4.0-17763-Microsoft #379-Microsoft Wed Mar 06 19:16:00 PST 2019 x86_64 GNU/Linux
```

cr1901, 2019-07-20:
@CarlFK Is that WSL? Note I have absolutely no experience with WSL :(.

CarlFK, 2019-07-23:
Yes, WSL.

no sad faces, most of my experience with WSL has been following these instructions, so we are in the same rocky boat.