timvideos/litex-buildenv #173: Conda's toolchain is unable to compile Linux

mateusz-holenko, 2019-07-15:
I'm working on adding support for `VexRiscv` target in `scripts/build-linux.sh` and noticed that it''s not possible to build upstream Linux kernel with the toolchain installed from Conda:

<pre>
litex-buildenv/build/conda/bin/../lib/gcc/riscv32-elf/8.2.0/../../../../riscv32-elf/bin/ld: -shared not supported
</pre>

mithro, 2019-07-15:
These are build here -> https://github.com/litex-hub/litex-conda

What needs to change there to make it work?

mateusz-holenko, 2019-07-16:
There is `--enable-shared` switch in `ld`'s configure script that could be added here: https://github.com/litex-hub/litex-conda/blob/master/binutils/build.sh#L78, but I don't know if this would solve all the issues. I haven't recompiled the whole package as it was quite cumbersome to do on my host machine -- is Travis configured for this repo in a way, that creating a PR would build all packages?

According to this: https://github.com/riscv/riscv-gnu-toolchain/issues/335#issuecomment-418530499 discussion, using newlib toolchain to build Linux kernel is discouraged anyway.

I'm just wondering if creating a new package with a toolchain built from this repository: https://github.com/riscv/riscv-gnu-toolchain wouldn't be a better idea. What do you think, @mithro?

mithro, 2019-07-16:
> There is `--enable-shared` switch in `ld`'s configure script that could be added here: https://github.com/litex-hub/litex-conda/blob/master/binutils/build.sh#L78, but I don't know if this would solve all the issues.

Attempting to build on https://github.com/litex-hub/litex-conda/pull/71

> I haven't recompiled the whole package as it was quite cumbersome to do on my host machine -- is Travis configured for this repo in a way, that creating a PR would build all packages?

Yes it would. You can also enable travis on your own fork.

> According to this: [riscv/riscv-gnu-toolchain#335 (comment)](https://github.com/riscv/riscv-gnu-toolchain/issues/335#issuecomment-418530499) discussion, using newlib toolchain to build Linux kernel is discouraged anyway.

Do we need a `gcc/linux` toolchain like we have `gcc/newlib` and `gcc/nostdc`?

> I'm just wondering if creating a new package with a toolchain built from this repository: https://github.com/riscv/riscv-gnu-toolchain wouldn't be a better idea. What do you think, @mithro?

We want to use the same upstream toolchain on lm32, or1k and riscv.

mateusz-holenko, 2019-07-18:
I made a mistake and `--enable-shared` does not add support for handling shared libraries to `ld`, but builds binutil's tools in a way they use shared versions of `libopcodes`/`libbfd` etc. libraries.

It explains the error on Travis CI when trying to call `as --version`.

I reproduced it locally and compared two compiled versions - one with `--enable-shared` and the second without it. The first one worked only after setting `LD_LIBRARY_PATH` to point to the folder with compiled shared binutil's libs (`x86_64-pc-linux-gnu/riscv32-unknown-elf/lib`; this folder is not present at all when compiling without `--enable-shared`).

mateusz-holenko, 2019-07-18:
> Do we need a gcc/linux toolchain like we have gcc/newlib and gcc/nostdc?

This source: https://github.com/michaeljclark/busybear-linux/issues/5#issuecomment-418509114 suggests that `gcc/glibc` should be used to build Linux as newer `newlib` does not support dynamic linking required by the kernel to handle `vdso.so` library.

> We want to use the same upstream toolchain on lm32, or1k and riscv.

Out of curiosity, is `or1k` supported by the upstream GCC at all?

mithro, 2019-07-18:
> I made a mistake and `--enable-shared` does not add support for handling shared libraries to `ld`, but builds binutil's tools in a way they use shared versions of `libopcodes`/`libbfd` etc. [...]

I don't think we need this behaviour?

> + I proposed to add this line to `binutils` package, but the toolchain seems to come with it's own `ld`: `conda/bin/riscv32-elf-ld` vs `conda/riscv32-elf/bin/ld`

Which line?

> > Do we need a gcc/linux toolchain like we have gcc/newlib and gcc/nostdc?
> 
> This source: [michaeljclark/busybear-linux#5 (comment)](https://github.com/michaeljclark/busybear-linux/issues/5#issuecomment-418509114) suggests that `gcc/glibc` should be used to build Linux as newer `newlib` does not support dynamic linking required by the kernel to handle `vdso.so` library.

Do you know if we use [musl](https://www.musl-libc.org/) rather than glibc? It is unclear to me if glibc supports or1k and riscv32 upstream yet?

@stffrdhrn - Do you know about or1k status in glibc?

> > We want to use the same upstream toolchain on lm32, or1k and riscv.
> 
> Out of curiosity, is `or1k` supported by the upstream GCC at all?

or1k - **Yes**! As of gcc-9.0 (?) thanks to the work of @stffrdhrn 
lm32 - I believe is in the process of going away. Unclear if it has happened yet.

stffrdhrn, 2019-07-18:
> 
> Do you know if we use [musl](https://www.musl-libc.org/) rather than glibc? It is unclear to me if glibc supports or1k and riscv32 upstream yet?
> 
> @stffrdhrn - Do you know about or1k status in glibc?

We have a port in this repo: https://github.com/openrisc/or1k-glibc.  But the port didn't go upstream due to no GCC.  I have never built it but I assume it will not be much work to finish it. Now that that GCC is upstream it's on my todo list.  But so far I have been using musl.

mithro, 2019-07-22:
`$SRC_DIR/musl/configure: unknown or unsupported target "riscv32-elf"` -- https://travis-ci.org/mithro/conda-hdmi2usb-packages/jobs/562208408

Does linux+musl work on `riscv32`?

mithro, 2019-07-22:
https://bellard.org/tinyemu/buildroot.html

mithro, 2019-07-22:
https://www.openwall.com/lists/musl/2018/09/28/4

mithro, 2019-07-22:
https://groups.google.com/a/groups.riscv.org/forum/#!topic/sw-dev/XHuc4hNqTj4

mateusz-holenko, 2019-07-23:
@mithro I'm working on this locally as well and it seems that the `musl` mainline has only support for `riscv64` that has been added by this commit: https://git.musl-libc.org/cgit/musl/commit/?id=0a48860c27a8eb291bcc7616ea9eb073dc660cab included in 1.1.23 release with further fixes (not yet included in any release).

mateusz-holenko, 2019-07-23:
I tried to build `gcc/musl` with musl's master for `riscv64-linux-musl` target using https://github.com/richfelker/musl-cross-make.git, but it failed with:

<pre>
In file included from ../src_musl/src/stat/fstatat.c:8:
../src_musl/src/stat/fstatat.c: In function ‘fstatat_statx’:
../src_musl/src/stat/fstatat.c:40:22: error: ‘SYS_statx’ undeclared (first use in this function); did you mean ‘SYS_statfs’?
  int ret = __syscall(SYS_statx, fd, path, flag, 0x7ff, &stx);
</pre>

mithro, 2019-07-23:
@mateusz-holenko I've upgraded binutils, gcc and newlib in conda (see https://travis-ci.com/litex-hub/litex-conda/builds/120073354).

mithro, 2019-07-23:
I have an open pull request to add Linux + musl toolchain at https://github.com/litex-hub/litex-conda/pull/75 but it doesn't work on either or1k or riscv32....

mithro, 2019-07-25:
@stffrdhrn - Do you know what is needed to build Linux + or1k + musl? https://github.com/litex-hub/litex-conda/pull/75 is failing with;
`PACKAGE=gcc/linux-musl TOOLCHAIN_ARCH=or1k` currently fails https://travis-ci.com/litex-hub/litex-conda/jobs/218353487
```
using compiler runtime libraries: -lgcc -lgcc_eh
checking whether compiler's long double definition matches float.h... no
$SRC_DIR/musl/configure: error: unsupported long double type
```

stffrdhrn, 2019-07-25:
Hello, I am not sure.  I have not seen this error before.  One thing looking strange is that it says "or1k-elf-gcc"?

For my builds of musl I use [musl-cross-make](https://github.com/stffrdhrn/musl-cross-make.git).

See: https://github.com/stffrdhrn/or1k-toolchain-build/blob/master/or1k-toolchain-build/build-gcc.sh#L134

```
or1k-elf-gcc (9.1.0) 9.1.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
---
checking for C compiler... $BUILD_PREFIX/bin/x86_64-conda_cos6-linux-gnu-cc
checking whether C compiler works... yes
checking whether compiler accepts -Werror=unknown-warning-option... no
checking whether compiler accepts -Werror=unused-command-line-argument... no
checking whether compiler accepts -Werror=ignored-optimization-argument... no
checking whether linker accepts -Werror=unknown-warning-option... no
checking whether linker accepts -Werror=unused-command-line-argument... no
checking for C compiler family... gcc
checking for toolchain wrapper to build... gcc
checking target system type... or1k-elf
checking whether compiler accepts -std=c99... yes
checking whether compiler accepts -nostdinc... yes
checking whether compiler accepts -ffreestanding... yes
checking whether compiler accepts -fexcess-precision=standard... yes
checking whether compiler accepts -frounding-math... yes
checking whether compiler needs attribute((may_alias)) suppression... no
checking whether compiler accepts -Wa,--noexecstack... yes
checking whether compiler accepts -fno-stack-protector... yes
checking whether compiler accepts -fno-tree-loop-distribute-patterns... yes
checking whether we should preprocess assembly to add debugging information... no
checking for optimization settings... using provided CFLAGS
checking whether compiler accepts -pipe... yes
checking whether compiler accepts -fomit-frame-pointer... yes
checking whether compiler accepts -fno-unwind-tables... yes
checking whether compiler accepts -fno-asynchronous-unwind-tables... yes
checking whether compiler accepts -ffunction-sections... yes
checking whether compiler accepts -fdata-sections... yes
checking whether compiler accepts -Werror=implicit-function-declaration... yes
checking whether compiler accepts -Werror=implicit-int... yes
checking whether compiler accepts -Werror=pointer-sign... yes
checking whether compiler accepts -Werror=pointer-arith... yes
checking preprocessor condition __PIC__... true
checking whether linker accepts -Wl,--sort-section,alignment... yes
checking whether linker accepts -Wl,--sort-common... yes
checking whether linker accepts -Wl,--gc-sections... yes
checking whether linker accepts -Wl,--hash-style=both... yes
checking whether linker accepts -Wl,--no-undefined... yes
checking whether linker accepts -Wl,--exclude-libs=ALL... yes
checking whether linker accepts -Wl,--dynamic-list=$SRC_DIR/musl/dynamic.list... yes
checking whether linker accepts -lgcc... yes
checking whether linker accepts -lgcc_eh... yes
using compiler runtime libraries: -lgcc -lgcc_eh
checking whether compiler's long double definition matches float.h... no
```

stffrdhrn, 2019-07-25:
Some more investigation:

Just googling for this error, I see the below about out of tree builds, not sure if its related.
https://www.openwall.com/lists/musl/2014/02/06/2

From the error I can see its a check in musl.  It checks the GCC `$BUILD_PREFIX/bin/x86_64-conda_cos6-linux-gnu-cc` has proper support for long double.
http://git.musl-libc.org/cgit/musl/tree/configure#n704

stffrdhrn, 2019-07-25:
OK, I can't seem to reproduce even with my or1k-elf toolchain with  `./configure --target=or1k-elf`. 

```
$ CC=or1k-elf-gcc-9.1.1 ./configure --target=or1k-elf

checking for C compiler... or1k-elf-gcc-9.1.1
checking whether C compiler works... yes
checking whether compiler accepts -Werror=unknown-warning-option... no
checking whether compiler accepts -Werror=unused-command-line-argument... no
checking whether compiler accepts -Werror=ignored-optimization-argument... no
checking whether linker accepts -Werror=unknown-warning-option... no
checking whether linker accepts -Werror=unused-command-line-argument... no
checking for C compiler family... gcc
checking for toolchain wrapper to build... none
checking target system type... or1k-elf
checking whether compiler accepts -std=c99... yes
checking whether compiler accepts -nostdinc... yes
checking whether compiler accepts -ffreestanding... yes
checking whether compiler accepts -fexcess-precision=standard... yes
checking whether compiler accepts -frounding-math... yes
checking whether compiler needs attribute((may_alias)) suppression... no
checking whether compiler accepts -Wa,--noexecstack... yes
checking whether compiler accepts -fno-stack-protector... yes
checking whether compiler accepts -fno-tree-loop-distribute-patterns... yes
checking whether we should preprocess assembly to add debugging information... no
checking for optimization settings... using defaults
checking whether compiler accepts -Os... yes
components to be optimized for speed: internal malloc string
checking whether compiler accepts -pipe... yes
checking whether compiler accepts -fomit-frame-pointer... yes
checking whether compiler accepts -fno-unwind-tables... yes
checking whether compiler accepts -fno-asynchronous-unwind-tables... yes
checking whether compiler accepts -ffunction-sections... yes
checking whether compiler accepts -fdata-sections... yes
checking whether compiler accepts -Werror=implicit-function-declaration... yes
checking whether compiler accepts -Werror=implicit-int... yes
checking whether compiler accepts -Werror=pointer-sign... yes
checking whether compiler accepts -Werror=pointer-arith... yes
checking preprocessor condition __PIC__... false
checking whether linker accepts -Wl,--sort-section,alignment... no
checking whether linker accepts -Wl,--sort-common... no
checking whether linker accepts -Wl,--gc-sections... no
checking whether linker accepts -Wl,--hash-style=both... no
checking whether linker accepts -Wl,--no-undefined... no
checking whether linker accepts -Wl,--exclude-libs=ALL... no
checking whether linker accepts -Wl,--dynamic-list=./dynamic.list... no
checking whether linker accepts -lgcc... no
checking whether linker accepts -lcompiler_rt... no
checking whether linker accepts /home/shorne/work/gnu-toolchain/local/lib/gcc/or1k-elf/9.1.1/libgcc.a... no
checking whether linker accepts libpcc.a... no
using compiler runtime libraries: 
checking whether compiler's long double definition matches float.h... yes
checking preprocessor condition __FAST_MATH__... false
creating config.mak... done
```

mithro, 2019-07-27:
Well I managed to get further now. However, I get;
```
cp obj/crt/Scrt1.o lib/Scrt1.o
cp obj/crt/crt1.o lib/crt1.o
cp obj/crt/rcrt1.o lib/rcrt1.o
/home/tansell/conda/conda-bld/gcc-or1k-elf-linux-musl_1564260628138/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_/bin/../lib/gcc/or1k-linux-musl/9.1.0/../../../../or1k-linux-musl/bin/ld: -shared not supported
collect2: error: ld returned 1 exit status
make: *** [Makefile:160: lib/libc.so] Error 1
make: *** Waiting for unfinished jobs....
Traceback (most recent call last):
  File "/home/tansell/conda/bin/conda-build", line 11, in <module>
    sys.exit(main())
  File "/home/tansell/conda/lib/python3.7/site-packages/conda_build/cli/main_build.py", line 445, in main
    execute(sys.argv[1:])
  File "/home/tansell/conda/lib/python3.7/site-packages/conda_build/cli/main_build.py", line 436, in execute
    verify=args.verify, variants=args.variants)
  File "/home/tansell/conda/lib/python3.7/site-packages/conda_build/api.py", line 209, in build
    notest=notest, need_source_download=need_source_download, variants=variants)
  File "/home/tansell/conda/lib/python3.7/site-packages/conda_build/build.py", line 2313, in build_tree
    notest=notest,
  File "/home/tansell/conda/lib/python3.7/site-packages/conda_build/build.py", line 1461, in build
    cwd=src_dir, stats=build_stats)
  File "/home/tansell/conda/lib/python3.7/site-packages/conda_build/utils.py", line 397, in check_call_env
    return _func_defaulting_env_to_os_environ('call', *popenargs, **kwargs)
  File "/home/tansell/conda/lib/python3.7/site-packages/conda_build/utils.py", line 377, in _func_defaulting_env_to_os_environ
    raise subprocess.CalledProcessError(proc.returncode, _args)
subprocess.CalledProcessError: Command '['/bin/bash', '-e', '/home/tansell/conda/conda-bld/gcc-or1k-elf-linux-musl_1564260628138/work/conda_build.sh']' returned non-zero exit status 2.
```

mithro, 2019-07-27:
So, it seems the root cause is that binutils for baremetal is different from binuils for linux....

mithro, 2019-07-27:
```
.								.
./bin								./bin
./bin/or1k-elf-addr2line				      |	./bin/or1k-linux-musl-addr2line
./bin/or1k-elf-ar					      |	./bin/or1k-linux-musl-ar
./bin/or1k-elf-as					      |	./bin/or1k-linux-musl-as
./bin/or1k-elf-c++filt					      |	./bin/or1k-linux-musl-c++filt
./bin/or1k-elf-elfedit					      |	./bin/or1k-linux-musl-elfedit
./bin/or1k-elf-gprof					      |	./bin/or1k-linux-musl-gprof
./bin/or1k-elf-ld					      |	./bin/or1k-linux-musl-ld
./bin/or1k-elf-ld.bfd					      |	./bin/or1k-linux-musl-ld.bfd
./bin/or1k-elf-nm					      |	./bin/or1k-linux-musl-nm
./bin/or1k-elf-objcopy					      |	./bin/or1k-linux-musl-objcopy
./bin/or1k-elf-objdump					      |	./bin/or1k-linux-musl-objdump
./bin/or1k-elf-ranlib					      |	./bin/or1k-linux-musl-ranlib
./bin/or1k-elf-readelf					      |	./bin/or1k-linux-musl-readelf
./bin/or1k-elf-size					      |	./bin/or1k-linux-musl-size
./bin/or1k-elf-strings					      |	./bin/or1k-linux-musl-strings
./bin/or1k-elf-strip					      |	./bin/or1k-linux-musl-strip
./or1k-elf						      |	./or1k-linux-musl
./or1k-elf/bin						      |	./or1k-linux-musl/bin
./or1k-elf/bin/ar					      |	./or1k-linux-musl/bin/ar
./or1k-elf/bin/as					      |	./or1k-linux-musl/bin/as
./or1k-elf/bin/ld					      |	./or1k-linux-musl/bin/ld
./or1k-elf/bin/ld.bfd					      |	./or1k-linux-musl/bin/ld.bfd
./or1k-elf/bin/nm					      |	./or1k-linux-musl/bin/nm
./or1k-elf/bin/objcopy					      |	./or1k-linux-musl/bin/objcopy
./or1k-elf/bin/objdump					      |	./or1k-linux-musl/bin/objdump
./or1k-elf/bin/ranlib					      |	./or1k-linux-musl/bin/ranlib
./or1k-elf/bin/readelf					      |	./or1k-linux-musl/bin/readelf
./or1k-elf/bin/strip					      |	./or1k-linux-musl/bin/strip
./or1k-elf/lib						      |	./or1k-linux-musl/lib
./or1k-elf/lib/ldscripts				      |	./or1k-linux-musl/lib/ldscripts
./or1k-elf/lib/ldscripts/elf32or1k.x			      |	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.x
./or1k-elf/lib/ldscripts/elf32or1k.xbn			      |	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xbn
./or1k-elf/lib/ldscripts/elf32or1k.xc			      |	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xc
./or1k-elf/lib/ldscripts/elf32or1k.xce			      |	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xce
./or1k-elf/lib/ldscripts/elf32or1k.xe			      |	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xd
./or1k-elf/lib/ldscripts/elf32or1k.xn			      |	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xdc
./or1k-elf/lib/ldscripts/elf32or1k.xr			      |	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xdce
./or1k-elf/lib/ldscripts/elf32or1k.xu			      |	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xde
./or1k-elf/lib/ldscripts/elf32or1k.xw			      |	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xdw
./or1k-elf/lib/ldscripts/elf32or1k.xwe			      |	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xdwe
							      >	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xe
							      >	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xn
							      >	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xr
							      >	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xs
							      >	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xsc
							      >	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xsce
							      >	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xse
							      >	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xsw
							      >	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xswe
							      >	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xu
							      >	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xw
							      >	./or1k-linux-musl/lib/ldscripts/elf32or1k_linux.xwe
./share								./share
./share/info							./share/info
./share/info/as.info						./share/info/as.info
./share/info/bfd.info						./share/info/bfd.info
./share/info/binutils.info					./share/info/binutils.info
./share/info/dir						./share/info/dir
./share/info/gprof.info						./share/info/gprof.info
./share/info/ld.info						./share/info/ld.info
./share/locale							./share/locale
./share/locale/bg						./share/locale/bg
./share/locale/bg/LC_MESSAGES					./share/locale/bg/LC_MESSAGES
./share/locale/bg/LC_MESSAGES/binutils.mo			./share/locale/bg/LC_MESSAGES/binutils.mo
./share/locale/bg/LC_MESSAGES/gprof.mo				./share/locale/bg/LC_MESSAGES/gprof.mo
./share/locale/bg/LC_MESSAGES/ld.mo				./share/locale/bg/LC_MESSAGES/ld.mo
./share/locale/ca						./share/locale/ca
[... 157 lines cut ...]
./share/locale/zh_TW						./share/locale/zh_TW
./share/locale/zh_TW/LC_MESSAGES				./share/locale/zh_TW/LC_MESSAGES
./share/locale/zh_TW/LC_MESSAGES/binutils.mo			./share/locale/zh_TW/LC_MESSAGES/binutils.mo
./share/locale/zh_TW/LC_MESSAGES/ld.mo				./share/locale/zh_TW/LC_MESSAGES/ld.mo
./share/man							./share/man
./share/man/man1						./share/man/man1
./share/man/man1/or1k-elf-addr2line.1			      |	./share/man/man1/or1k-linux-musl-addr2line.1
./share/man/man1/or1k-elf-ar.1				      |	./share/man/man1/or1k-linux-musl-ar.1
./share/man/man1/or1k-elf-as.1				      |	./share/man/man1/or1k-linux-musl-as.1
./share/man/man1/or1k-elf-c++filt.1			      |	./share/man/man1/or1k-linux-musl-c++filt.1
./share/man/man1/or1k-elf-dlltool.1			      |	./share/man/man1/or1k-linux-musl-dlltool.1
./share/man/man1/or1k-elf-elfedit.1			      |	./share/man/man1/or1k-linux-musl-elfedit.1
./share/man/man1/or1k-elf-gprof.1			      |	./share/man/man1/or1k-linux-musl-gprof.1
./share/man/man1/or1k-elf-ld.1				      |	./share/man/man1/or1k-linux-musl-ld.1
[... cut ...]
./share/man/man1/or1k-elf-readelf.1			      |	./share/man/man1/or1k-linux-musl-readelf.1
./share/man/man1/or1k-elf-size.1			      |	./share/man/man1/or1k-linux-musl-size.1
./share/man/man1/or1k-elf-strings.1			      |	./share/man/man1/or1k-linux-musl-strings.1
./share/man/man1/or1k-elf-strip.1			      |	./share/man/man1/or1k-linux-musl-strip.1
./share/man/man1/or1k-elf-windmc.1			      |	./share/man/man1/or1k-linux-musl-windmc.1
./share/man/man1/or1k-elf-windres.1			      |	./share/man/man1/or1k-linux-musl-windres.1
```

stffrdhrn, 2019-07-28:
Ok. It looked strange mixing baremetal and musl toolchains.  Let me know if you need help with riscv too

mithro, 2019-07-28:
@stffrdhrn - The gcc without a standard libc should be equivalent to the "stage 1" compiler.

mithro, 2019-07-28:
@stffrdhrn - So I was trying to prevent having to rebuild the "stage 1" by using a prebuilt "stage 1" compiler, however I just can't seem to make it work so I'm resorting to doing a more normal 2-stage GCC compile.

mithro, 2019-07-28:
```
/home/tansell/conda/conda-bld/gcc-or1k-elf-linux-musl_1564336611242/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_/bin/../lib/gcc/or1k-linux-musl/9.1.0/../../../../or1k-linux-musl/bin/ld: obj/src/complex/__cexp.lo: in function `__ldexp_cexp':
__cexp.c:(.text.__ldexp_cexp+0x60): undefined reference to `__subdf3'
/home/tansell/conda/conda-bld/gcc-or1k-elf-linux-musl_1564336611242/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_/bin/../lib/gcc/or1k-linux-musl/9.1.0/../../../../or1k-linux-musl/bin/ld: __cexp.c:(.text.__ldexp_cexp+0xdc): undefined reference to `__muldf3'
/home/tansell/conda/conda-bld/gcc-or1k-elf-linux-musl_1564336611242/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_/bin/../lib/gcc/or1k-linux-musl/9.1.0/../../../../or1k-linux-musl/bin/ld: __cexp.c:(.text.__ldexp_cexp+0xf0): undefined reference to `__muldf3'
/home/tansell/conda/conda-bld/gcc-or1k-elf-linux-musl_1564336611242/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_/bin/../lib/gcc/or1k-linux-musl/9.1.0/../../../../or1k-linux-musl/bin/ld: __cexp.c:(.text.__ldexp_cexp+0x104): undefined reference to `__muldf3'
/home/tansell/conda/conda-bld/gcc-or1k-elf-linux-musl_1564336611242/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_/bin/../lib/gcc/or1k-linux-musl/9.1.0/../../../../or1k-linux-musl/bin/ld: __cexp.c:(.text.__ldexp_cexp+0x12c): undefined reference to `__muldf3'
/home/tansell/conda/conda-bld/gcc-or1k-elf-linux-musl_1564336611242/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_/bin/../lib/gcc/or1k-linux-musl/9.1.0/../../../../or1k-linux-musl/bin/ld: __cexp.c:(.text.__ldexp_cexp+0x140): undefined reference to `__muldf3'
/home/tansell/conda/conda-bld/gcc-or1k-elf-linux-musl_1564336611242/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_/bin/../lib/gcc/or1k-linux-musl/9.1.0/../../../../or1k-linux-musl/bin/ld: obj/src/complex/__cexp.lo:__cexp.c:(.text.__ldexp_cexp+0x154): more undefined references to `__muldf3' follow
/home/tansell/conda/conda-bld/gcc-or1k-elf-linux-musl_1564336611242/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_/bin/../lib/gcc/or1k-linux-musl/9.1.0/../../../../or1k-linux-musl/bin/ld: obj/src/complex/__cexpf.lo: in function `__ldexp_cexpf':
__cexpf.c:(.text.__ldexp_cexpf+0x40): undefined reference to `__subsf3'
```

mithro, 2019-07-28:
The issue seems to be that musl doesn't support multilib, trying without it.

mithro, 2019-07-30:
I took a shot at using musl-cross-make as it seems like upstream musl doesn't have riscv32 support. See https://github.com/litex-hub/litex-conda/pull/80/files and https://travis-ci.org/mithro/conda-hdmi2usb-packages/builds/565270987 -- seems to build and pass an initial smoke test....

https://git.zv.io/toolchains/musl-cross-make/blob/musl-git/scripts/gen_riscv_patch

mithro, 2019-07-30:
I got a successful build of musl + or1k with the other approach at https://travis-ci.org/mithro/conda-hdmi2usb-packages/builds/565050257

mithro, 2019-07-31:
I have merged the pull requires which add conda packages for riscv32 with musl. @mateusz-holenko Can you see if using either of these toolchains enables riscv32 kernel to be built now?

mithro, 2019-08-08:
@mateusz-holenko - ping?

mateusz-holenko, 2019-08-12:
@mithro Do you mean https://github.com/timvideos/litex-buildenv/pull/188? As mentioned in the comment (https://github.com/timvideos/litex-buildenv/pull/188#issuecomment-518143532), I confirm that the current master (that uses `riscv32-linux-musl-gcc`) is able to build Linux kernel.

Sorry for not posting the status here, I must have focused on the PR and missed this thread.

mithro, 2019-08-12:
@mateusz-holenko - Great! Closing this now then.

closed 2019-08-12 by mithro