timvideos/litex-buildenv #193: Zephyr build is broken

mithro, 2019-08-08:
```
=== Initialized. Now run "west update" inside /home/travis/build/timvideos/litex-buildenv/third_party/zephyr.
+west build -b litex_vexriscv /home/travis/build/timvideos/litex-buildenv/third_party/zephyr/zephyr/samples/subsys/shell/shell_module --build-dir /home/travis/build/timvideos/litex-buildenv/build/icebreaker_base_vexriscv.lite//software/zephyr
-- west build: build configuration:
       source directory: /home/travis/build/timvideos/litex-buildenv/third_party/zephyr/zephyr/samples/subsys/shell/shell_module
       build directory: /home/travis/build/timvideos/litex-buildenv/build/icebreaker_base_vexriscv.lite/software/zephyr (created)
       BOARD: litex_vexriscv (origin: command line)
-- west build: generating a build system
Zephyr version: 1.14.99
-- Found PythonInterp: /home/travis/build/timvideos/litex-buildenv/build/conda/bin/python (found suitable version "3.7.0", minimum required is "3.4") 
-- Selected BOARD litex_vexriscv
CMake Error at /home/travis/build/timvideos/litex-buildenv/third_party/zephyr/zephyr/cmake/toolchain/zephyr/host-tools.cmake:39 (message):
  The SDK version you are using is too old, please update your SDK.
  You need at least SDK version 0.10.2.
  You have version 0.10.1
  (/home/travis/build/timvideos/litex-buildenv/build/zephyr_sdk).
  The new version of the SDK can be downloaded from:
  
  https://github.com/zephyrproject-rtos/sdk-ng/releases/download/v0.10.2/zephyr-sdk-0.10.2-setup.run
Call Stack (most recent call first):
  /home/travis/build/timvideos/litex-buildenv/third_party/zephyr/zephyr/cmake/host-tools.cmake:3 (include)
  /home/travis/build/timvideos/litex-buildenv/third_party/zephyr/zephyr/cmake/app/boilerplate.cmake:414 (include)
  CMakeLists.txt:4 (include)
-- Configuring incomplete, errors occurred!
ERROR: command exited with status 1: /home/travis/build/timvideos/litex-buildenv/build/conda/bin/cmake -B/home/travis/build/timvideos/litex-buildenv/build/icebreaker_base_vexriscv.lite/software/zephyr -S/home/travis/build/timvideos/litex-buildenv/third_party/zephyr/zephyr/samples/subsys/shell/shell_module -GNinja -DBOARD=litex_vexriscv
Errors from failures...
===========================================
```

mithro, 2019-08-08:
@mateusz-holenko - Can you please take a look?

mateusz-holenko, 2019-08-12:
@mithro It looks like a new Zephyr SDK has been released and we need to bump it in scripts. I'll check it in a moment.

PR #195 (2019-08-12): 193 broken zephyr
This fixes #193.
closed 2019-08-12 by mithro