timvideos/litex-buildenv #194: ./scripts/download-env-root.sh doesn't exists

FrankBuss, 2019-08-09:
I was following the instructions at https://github.com/timvideos/litex-buildenv/blob/master/README.md but the file ./scripts/download-env-root.sh doesn't exists, only download-env.sh.

mithro, 2019-08-09:
Looks like our documentation is out of date. The `download-env.sh` script has become https://github.com/timvideos/litex-buildenv/blob/master/scripts/debian-setup.sh

FrankBuss, 2019-08-10:
This fails:
```
E: Failed to fetch http://ppa.launchpad.net/obsproject/obs-studio/ubuntu/dists/eoan/main/binary-i386/Packages  404  Not Found
E: Failed to fetch http://ppa.launchpad.net/timvideos/fpga-support/ubuntu/dists/eoan/main/binary-amd64/Packages  404  Not Found
```

mithro, 2019-08-10:
Seems we don't have eoan support. Try skipping this step?

closed 2019-09-02 by mithro

mithro, 2019-09-04:
This should now be fixed! Sorry about taking so long.