timvideos/litex-buildenv #196: Supporting the MATRIX Voice (xc6slx9-2ftg256c w/ MT47H32M16 DDR2 RAM)

samreenislam, 2019-08-15:
Label: type-new-project

Hello,

I'm trying to support the MATRIX Voice (xc6slx9-2ftg256c w/ MT47H32M16 DDR2 RAM, mx25l6406e SPI Flash) and deploy an lm32 CPU on its Spartan-6 to run Micropython using your project.

I created a matrix_voice.py file with the constraints in the "platforms" folder and base.py & Makefile.mk in the "targets" folder.

I'm able to enter the environment using
```
source scripts/enter-env.sh
```
After that, much of the content that would go in targets/matrix_voice/base.py for the MATRIX Voice have already been written in Verilog. 

Do you know of any resources to use Migen & LiteX alongside pre-existing Verilog for things like "IDDR2", "ODDR2", "BUFG", etc? I would appreciate if someone could point me in the right direction.

Thank you!

mithro, 2019-08-15:
@samreenislam - Have you gotten to the stage of a LiteX BIOS prompt?

mithro, 2019-08-15:
@samreenislam FYI You can find us on irc.freenode.net on #litex & #timvideos IRC channels if you want to ask questions interactively.

mithro, 2019-08-15:
Some good examples of including / using Verilog files can be found at the CPU cores at https://github.com/enjoy-digital/litex/tree/master/litex/soc/cores/cpu

samreenislam, 2019-08-15:
> @samreenislam - Have you gotten to the stage of a LiteX BIOS prompt?

Yes, I have this right now
[image]

Struggling with the `make gateware` since I'm just learning Migen syntax :sweat_smile:

mithro, 2019-08-15:
That is the *bash* prompt. The LiteX BIOS prompt comes when you do a `make firmware-connect` and are successfully talking serial to the BIOS running on the softcore inside the FPGA.

mithro, 2019-08-15:
Some other examples here -> https://github.com/timvideos/litex-buildenv/wiki/LiteX-for-Hardware-Engineers#instances

samreenislam, 2019-08-15:
Oops, yeah not there yet. This is how far I've gotten. I need to fix my /targets/matrix_voice/base.py file. Thank you for the resources! 

I'll be sure to ping on the IRC channels if I need some more direction :smile: