timvideos/litex-buildenv #199: Conda failed to build hexfile

shenki, 2019-09-16:
On Debian testing with litex-buildenv v0.0.4-592-gf2c790cc36d7. I am configured for CPU=vexriscv PLATFORM=arty TARGET=net and am running download-env.sh for the first time:

```
Installing hexfile (python module)
Collecting git+https://github.com/mithro/hexfile.git
  Cloning https://github.com/mithro/hexfile.git to /tmp/pip-req-build-obrvob2z
  Running command git clone -q https://github.com/mithro/hexfile.git /tmp/pip-req-build-obrvob2z
Building wheels for collected packages: hexfile
  Building wheel for hexfile (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/joel/dev/litex/litex-buildenv/build/conda/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-obrvob2z/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-obrvob2z/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-7v59a8zz --python-tag cp37
       cwd: /tmp/pip-req-build-obrvob2z/
  Complete output (46 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib
  creating build/lib/hexfile
  copying hexfile/__init__.py -> build/lib/hexfile
  copying hexfile/core.py -> build/lib/hexfile
  installing to build/bdist.linux-x86_64/wheel
  running install
  running install_lib
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-req-build-obrvob2z/setup.py", line 30, in <module>
      url='http://www.github.com/ryansturmer/hexfile')
    File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 145, in setup
      return distutils.core.setup(**attrs)
    File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/dist.py", line 966, in run_commands
      self.run_command(cmd)
    File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/site-packages/wheel/bdist_wheel.py", line 228, in run
      self.run_command('install')
    File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 61, in run
      return orig.install.run(self)
    File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/command/install.py", line 557, in run
      self.run_command(cmd_name)
    File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/dist.py", line 984, in run_command
      cmd_obj.ensure_finalized()
    File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/cmd.py", line 107, in ensure_finalized
      self.finalize_options()
    File "/usr/lib/python3/dist-packages/setuptools/command/install_lib.py", line 17, in finalize_options
      self.set_undefined_options('install',('install_layout','install_layout'))
    File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/cmd.py", line 290, in set_undefined_options
      setattr(self, dst_option, getattr(src_cmd_obj, src_option))
    File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/cmd.py", line 103, in __getattr__
      raise AttributeError(attr)
  AttributeError: install_layout
  ----------------------------------------
  ERROR: Failed building wheel for hexfile
  Running setup.py clean for hexfile
Failed to build hexfile
Installing collected packages: hexfile
  Running setup.py install for hexfile ... error
    ERROR: Command errored out with exit status 1:
     command: /home/joel/dev/litex/litex-buildenv/build/conda/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-obrvob2z/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-obrvob2z/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-_l66kmmv/install-record.txt --single-version-externally-managed --compile
         cwd: /tmp/pip-req-build-obrvob2z/
    Complete output (38 lines):
    running install
    running build
    running build_py
    creating build
    creating build/lib
    creating build/lib/hexfile
    copying hexfile/__init__.py -> build/lib/hexfile
    copying hexfile/core.py -> build/lib/hexfile
    running install_lib
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-obrvob2z/setup.py", line 30, in <module>
        url='http://www.github.com/ryansturmer/hexfile')
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 145, in setup
        return distutils.core.setup(**attrs)
      File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/dist.py", line 966, in run_commands
        self.run_command(cmd)
      File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 61, in run
        return orig.install.run(self)
      File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/command/install.py", line 557, in run
        self.run_command(cmd_name)
      File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/dist.py", line 984, in run_command
        cmd_obj.ensure_finalized()
      File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/cmd.py", line 107, in ensure_finalized
        self.finalize_options()
      File "/usr/lib/python3/dist-packages/setuptools/command/install_lib.py", line 17, in finalize_options
        self.set_undefined_options('install',('install_layout','install_layout'))
      File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/cmd.py", line 290, in set_undefined_options
        setattr(self, dst_option, getattr(src_cmd_obj, src_option))
      File "/home/joel/dev/litex/litex-buildenv/build/conda/lib/python3.7/distutils/cmd.py", line 103, in __getattr__
        raise AttributeError(attr)
    AttributeError: install_layout
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/joel/dev/litex/litex-buildenv/build/conda/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-obrvob2z/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-obrvob2z/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-_l66kmmv/install-record.txt --single-version-externally-managed --compile Check the logs for full command output.
```

shenki, 2019-09-16:
I tried using pip to update setuptools in the conda env and it said:
```
Requirement already up-to-date: setuptools in /usr/lib/python3/dist-packages (41.2.0)
```

shenki, 2019-09-16:
I worked around the issue by removing python3-setuptools from my system.

mithro, 2019-09-28:
This is very weird... 

Nothing in the build environment should be using the system environment.

Any chance you could reproduce in a docker container so we can figure out what is going on?

shenki, 2019-09-28:
I couldn't reproduce inside a container.

```
docker run -it debian:testing /bin/bash
apt update && apt install git python3-setuptools wget make
adduser test
cd ~
git clone https://github.com/timvideos/litex-buildenv
cd litex-buildenv
export CPU=vexriscv PLATFORM=arty TARGET=net
./scripts/download-env.sh
```

Perhaps there was something else in my environment that made it break.

mithro, 2019-09-28:
Did you have some type of `PYTHONPATH`?

shenki, 2019-09-29:
I didn't have PYTHONPATH set, but I did have a copy of colorama in ~/.local. I assume this was there when I saw the failure.

shenki, 2019-09-29:
...and a copy of easy_install in ~/.local/bin.