timvideos/litex-buildenv #255: Add guides for NeTV2 and Pano Logic G2 to the Wiki

piotr-binkowski, 2019-12-03:
I don't see any way to create a PR for wiki changes so I decided to create a patch from my changes that can be reviewed and then merged
[0001-add-howtos-for-pano-logic-g2-and-netv2.patch.txt](https://github.com/timvideos/litex-buildenv/files/3916410/0001-add-howtos-for-pano-logic-g2-and-netv2.patch.txt)

mithro, 2019-12-03:
You should just be able to edit the wiki directly?

piotr-binkowski, 2019-12-04:
I just wasn't sure if I should edit it directly, I added those HowTos to the wiki
* https://github.com/timvideos/litex-buildenv/wiki/HowTo-Linux-on-Pano-Logic-G2
* https://github.com/timvideos/litex-buildenv/wiki/HowTo-HDMI2PCIe-on-NeTV2
and linked to them in [HowTos page](https://github.com/timvideos/litex-buildenv/wiki/HowTos)

closed 2019-12-04 by piotr-binkowski