timvideos/litex-buildenv #281: Conda Yosys package update

mmicko, 2020-01-05:
Please do update Yosys conda package, many features have been added for Xilinx flow to make synth for series 7 and Spartan 6 to work properly.

Adding this as issue as requested at https://twitter.com/mithro/status/1213869098251124737

Note how to build with Yosys added at (just search Yosys): 

https://github.com/timvideos/litex-buildenv/wiki/HowTo-LCA2018-FPGA-Miniconf-VexRiscv-Renode

But not yet valid when use with current conda package (fails on memory check on real hardware).

mithro, 2020-01-06:
The Yosys tests failing is currently blocking this update - https://travis-ci.com/mithro/litex-conda/jobs/272610002 and https://github.com/YosysHQ/yosys/issues/1615

mithro, 2020-01-08:
I added a patch to disable the failing test for now.

mmicko, 2020-01-08:
Thanks, we are looking for actual source of non-determinism in this case, will inform you when tests are passing again.

mithro, 2020-01-08:
New Yosys package should be available shortly. Can watch the build status at https://travis-ci.com/timvideos/litex-conda

@mmicko - Would appreciate some testing to see if it works.

PS The Yosys version was much newer than it might have seen because the v0.9 tag was not correctly pushed to the [timvideos/yosys](http://github.com/timvideos/yosys) fork and hence the git-describe output said v0.8 but was actually using something much closer to head (IE v0.9++).

mithro, 2020-01-08:
[linux-64/yosys-0.9_3002_g823a08e0-20200106_234044.tar.bz2](https://anaconda.org/TimVideos/yosys/files) is uploaded.

mmicko, 2020-01-08:
Thanks for this.
Did check with CPU=vexriscv PLATFORM=arty TARGET=net
And run : make gateware MAKE_LITEX_EXTRA_CMDLINE="--build-option synth_mode yosys"
that now gives working bitstream

mmicko, 2020-01-08:
Seams that valid would be:
MAKE_LITEX_EXTRA_CMDLINE="--build-option synth_mode yosys" make gateware 

otherwise some other things got changed, this enables building linux capable CPU. Will update on Wiki

mithro, 2020-01-08:
@mmicko - We should add that to the Travis CI to make sure it continues to work. We will need to figure out how to differentiate between Yosys for synthesis and Vivado for synthesis output.

mmicko, 2020-01-14:
@mithro Sorry for late response, but noticed a regression that appeared in that specific build you are using with conda package, so please do update yosys conda package to  https://github.com/YosysHQ/yosys/commit/9fbeb57bbdb98265b541d7a62213e83de63c8a1a in order to get valid bitstream produced by Vivado (reasons are explained in https://github.com/YosysHQ/yosys/pull/1623 )

mithro, 2020-01-15:
```
3.46s eda/syn/yosys: -  Package uploading...
 -------------------------------------------------------------------
Using Anaconda API: https://api.anaconda.org
Using "timvideos" as upload username
Processing '/tmp/really-really-really-really-really-really-really-really-really-really-really-really-really-long-path/conda/conda-bld/linux-64/yosys-0.9_3065_gabba1541-20200115_040235.tar.bz2'
Detecting file type...
File type is "conda"
Extracting conda package attributes for upload
Creating package "yosys"
Creating release "0.9_3065_gabba1541"
Uploading file "timvideos/yosys/0.9_3065_gabba1541/linux-64/yosys-0.9_3065_gabba1541-20200115_040235.tar.bz2"
 uploaded 10566 of 10566Kb: 100.00% ETA: 0.0 minutes
Upload complete
conda package located at:
https://anaconda.org/timvideos/yosys
```

mmicko, 2020-01-15:
Many thanks for this. Working fine now, and was able to boot to Linux. 
Will create a new wiki page for this flow. Have done multiple tests to confirm behavior.
Will post a link on it when I finish it.

closed 2020-01-15 by mmicko

mmicko, 2020-01-15:
Posting link to wiki page with new instructions : https://github.com/timvideos/litex-buildenv/wiki/HowTo-Yosys-Vivado-FPGA-Flow

mithro, 2020-01-15:
@mmicko Thanks!