timvideos/litex-buildenv #282: TinyFPGA BX: BIOS/Micropython does not start

ewenmcneill, 2020-01-08:
On TinyFPGA BX, BIOS/MicroPython serial does not start with current `litex-buildenv` and conda tools.

Based on `git` `bisect` it looks like the last working commit for TinyFPGA BX BIOS/MicroPython was da7488c2ab5e3d496ba4ecb5e482458302fb1c15, and first commit that builds which doesn't produce serial output is e4c6ffa96f6ac4048181363f9a9be2dbea4b19c3.  The main commit change in that range is 3a9127d08d8d0be9a68331bd71bf43fc216dbb26, which is a whole bunch of submodule updates to more recent Litex.  So presumably something changed around the gateware / CPU which needs more changes than were made to compensate for them.  Possibly related to CPU memory regions/addresses?

IIRC the TinyFPGA needs execute in place to work due to fairly constrained ice40, so possibly something related to execute in place got broken?

Testing with:

```
git checkout ....
make clean
make gateware
scripts/build-micropython.sh
```

FTR, notes from IRC:

```
12:38 < ewen> mithro: Looks like TinyFPGA BX / MicroPython serial/boot doesn't work after 3a9127d08d8d0be9a68331bd71bf43fc216dbb26
12:39 < ewen> mithro: That commit and the next one don't build due to API conflicts (that's a Litex submodule update set).
12:41 < ewen> mithro: Commits that do build after that don't show Litex BIOS or anything on serial
12:41 < ewen> (Looks like there are 71 commits after that.)
12:54 < ewen> mithro: e4c6ffa96f6ac4048181363f9a9be2dbea4b19c3 which is the next one that builds for me, doesn't work (no BIOS/serial console messages).
12:58 < ewen> FTR da7488c2ab5e3d496ba4ecb5e482458302fb1c15 is the last working commit that builds, and gives BIOS/serial.
12:59 < ewen> So my assumption is there's something in the (huge) litex update which broke TinyFPGA BX, and wasn't sufficiently compensated.
```

mithro, 2020-01-08:
I think the issue is the same as https://github.com/timvideos/litex-buildenv/issues/277

What hardware do you have with you?

ewenmcneill, 2020-01-08:
It's possible that it's the same issue as #277, but it's hard to be sure as the FPGA board and the CPU are different.

I have a [TinyFPGA BX](https://tinyfpga.com/bx/guide.html), with an [Adafruit USB Serial](https://www.adafruit.com/product/954) adapter for the console, connected as per https://github.com/timvideos/litex-buildenv/wiki/HowTo-FuPy-on-iCE40-Boards (which as noted above works on older git revisions, but not on newer git revisions, of litex-buildenv).

I'm using:

```
CPU=lm32 
CPU_VARIANT=minimal
PLATFORM=tinyfpga_bx
FIRMWARE=micropython
TARGET=base 
COMM_PORT=/dev/ttyUSB0
```

which was, historically, the best tested build of MicroPython for the ice40 chips (ie, LM32).  #277 was RISC-V, which I personally haven't seen working in litex-buildenv with MicroPython (I know xobs had it working in his build though).

I've got an iceBreaker here with me too (at the HackFest), so I was about to try that and see if that works/has same symptoms with the same revisions.

Ewen

ewenmcneill, 2020-01-08:
iceBreaker appears to work on latest `litex-buildenv` `git` (as well as the git rev working on TinyFPGA BX; the first broken one on the TinyFPGA BX doesn't build on iceBreaker for the same Litex API reasons).

So my working assumption now is that something else got fixed for iceBreaker that didn't also get fixed for the TinyFPGA BX / HX8K dev board.  In which case possibly #277 and this issue have a similar root cause.

Ewen

```
CPU=lm32 
CPU_VARIANT=minimal
PLATFORM=icebreaker
FIRMWARE=micropython
TARGET=base 
```

```
>>> 
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|

 (c) Copyright 2012-2019 Enjoy-Digital

 BIOS built on Jan  8 2020 14:36:43
 BIOS CRC passed (b9a5027a)

 Migen git sha1: 94db729
 LiteX git sha1: 02bfda5e

--=============== SoC ==================--
CPU:       LM32 @ 12MHz
ROM:       32KB
SRAM:      128KB

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from flash...
Executing booted program at 0x20028008

--============= Liftoff! ===============--
MicroPython v1.9.4-534-gd2bd40498-dirty on 2020-01-08; litex with lm32
>>> 
```

```
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/src/fpga/litex-buildenv$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/src/fpga/litex-buildenv$ git log -1 | head -1
commit be809038ea75a313e8b6bb236c7643a51c1526fb
(LX P=icebreaker.minimal F=micropython) ewen@parthenon:/src/fpga/litex-buildenv$ 
```

ewenmcneill, 2020-01-08:
FTR, based on IRC discussion, @mithro suggested testing on the Arty A7.  The Arty A7 *does* work, even on the latest `litex-buildenv` to run MicroPython.

Ewen

```
flterm --port=/dev/ttyUSB1 --kernel=build/arty_base_lm32//software/micropython/firmware.bin --speed=115200
[FLTERM] v2.4-29-g47d3b15 Starting...

litex> 
litex> 
litex> 
litex> 
litex> 
litex> 
litex> 
litex> 
litex> 
litex> exit
Command not found
litex> help
LiteX BIOS, available commands:
mr         - read address space
mw         - write address space
mc         - copy address space
fe         - erase whole flash
fw         - write to flash

crc        - compute CRC32 of a part of the address space
ident      - display identifier

reboot     - reset processor
serialboot - boot via SFL
flashboot  - boot from flash

memtest    - run a memory test
litex> reboot

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|

 (c) Copyright 2012-2019 Enjoy-Digital

 BIOS built on Jan  8 2020 16:17:17
 BIOS CRC passed (67da56c0)

 Migen git sha1: 94db729
 LiteX git sha1: 02bfda5e

--=============== SoC ==================--
CPU:       LM32 @ 100MHz
ROM:       32KB
SRAM:      32KB
L2:        8KB
MAIN-RAM:  262144KB

--========== Initialization ============--
Initializing SDRAM...
SDRAM now under software control
Read leveling:
m0, b0: |11111111111100000000000000000000| delays: 06+-06
m0, b1: |00000000000000011111111111100000| delays: 21+-06
m0, b2: |00000000000000000000000000000001| delays: 31+-00
m0, b3: |00000000000000000000000000000000| delays: -
m0, b4: |00000000000000000000000000000000| delays: -
m0, b5: |00000000000000000000000000000000| delays: -
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |00000000000000000000000000000000| delays: -
best: m0, b0 delays: 06+-06
m1, b0: |11111111111100000000000000000000| delays: 06+-06
m1, b1: |00000000000000011111111111100000| delays: 21+-06
m1, b2: |00000000000000000000000000000011| delays: 31+-01
m1, b3: |00000000000000000000000000000000| delays: -
m1, b4: |00000000000000000000000000000000| delays: -
m1, b5: |00000000000000000000000000000000| delays: -
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |00000000000000000000000000000000| delays: -
best: m1, b0 delays: 06+-06
SDRAM now under hardware control
Memtest OK

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[FLTERM] Received firmware download request from the device.
[FLTERM] Uploading kernel (245076 bytes)...
[FLTERM] Upload complete (7.7KB/s).
[FLTERM] Booting the device.
[FLTERM] Done.
Executing booted program at 0x40000000

--============= Liftoff! ===============--
MicroPython v1.9.4-534-gd2bd40498-dirty on 2020-01-08; litex with lm32
>>> 
```

niklasnisbeth, 2020-01-08:
I'm the reporter of #277 and FWIW I've arrived at the same revisions for when the breakage gets introduced so I can only assume they are related.

We were able to get a VexRiscV a bit further along the boot process (but not into a working firmware) by mapping SRAM to address 0x0, but now I am not sure if that is related at all.

I tried making a minimal project that depended on LiteX directly only (not buildenv) and it appears to have the same issues (as expected), but I have not yet found the exact commit where things break. I think that might be a better path forward than manually going through the giant merge commit that introduces the issue.

ewenmcneill, 2020-01-09:
Based on it working on some boards and not others, it seems at least plausible that it's related to changes in the SPI flash handling (one of @mithro's theories), especially since some of the platforms with problems are doing execute-in-place (XIP) from the SPI flash -- so the SPI flash access not working will basically prevent the CPU from executing *anything*.

But yes, it seems like this and #277 probably have the same root cause.  Somewhere in the bunch of LiteX changes dragged in a couple of months ago.   Reproducing direction on LiteX (without buildenv) seems like a useful way to potentially narrow down the cause.   The other option that comes to mind is to speculatively try rolling back the LiteX SPI code and see if that helps.   (Although IIRC there were a bunch of CSR changes in that time frame too, which perhaps makes everything more interdependent.)

Ewen

ewenmcneill, 2020-02-10:
Closing issue in favour of tracking this regression (apparently in the iCE40 boards that rely on execute-in-place from SPI) in https://github.com/timvideos/litex-buildenv/issues/277, which was opened first.

Ewen

closed 2020-02-10 by ewenmcneill