timvideos/litex-buildenv #319: missing libtinfo.so.5 causes Vivado install to hang

pdp7, 2020-02-02:
when running:
```
make gateware
```
I get this error:
```
application-specific initialization failed: couldn't load file "librdi_commontasks.so": libtinfo.so.5: cannot open shared object file: No such file or directory
```

pdp7, 2020-02-02:
Resolution:
I had to install: `libtinfo5`

mithro, 2020-02-03:
The `download-env.sh` script should check that the `libtinfo5` library is available so that Vivado doesn't fails to find this library and just gets stuck without really doing nothing.

CarlFK, 2020-02-22:
```
sudo apt install libtinfo-dev
sudo ln -s /lib/x86_64-linux-gnu/libtinfo.so.6 /lib/x86_64-linux-gnu/libtinfo.so.5
```
found this here:

https://askubuntu.com/questions/1155936/how-do-i-uninstall-vivado-on-ubuntu-19-04/1193020#1193020