timvideos/litex-buildenv #320: HowTo refers to red highlight text

pdp7, 2020-02-02:
[This HowTo](https://github.com/timvideos/litex-buildenv/wiki/HowTo-LCA2018-FPGA-Miniconf-VexRiscv-Renode) references red text 5 times but there is no color formatting.

Example:
> Please make sure the number shown in red above is less than 15 minutes!

mithro, 2020-02-03:
PTAL - @mateusz-holenko

mithro, 2020-02-03:
I wonder if this means we really should be using sphinx + readthedocs instead of github's wiki?

mateusz-holenko, 2020-02-19:
As mentioned at the top of the document:

> This is a modified version of the [Linux.conf.au 2018 FPGA Miniconf tutorial](https://github.com/timvideos/litex-buildenv/wiki/HowTo-LCA2018-FPGA-Miniconf)  that targets VexRiscv CPU (instead of mor1kx) and Renode (instead of qemu).

and the color-formatting references are simpy copied from the linked tutorial.

@mithro - do you know where the original version of this tutorial is, so that I can check what was coloured as red and update both documents on GitHub's wiki?

mithro, 2020-02-21:
It was in the Google Doc at https://docs.google.com/document/d/1b2rBju9Fxjlcv6y3Ib1WcrvmzPA63Pe-pnBaziJXZNY/edit

Look in the revision history and look at the revision named "Before Delete / Move"

mateusz-holenko, 2020-02-24:
@mithro: I'm afraid I have no access to the history of this document - the menu option for revision history is just greyed-out.

mithro, 2020-02-25:
@mateusz-holenko -- I shared a copy of the old version of the document directly with you.

mateusz-holenko, 2020-02-25:
Thanks @mithro -I removed all references to the text in red by rephrasing some fragments of the tutorial (in both versions).

BTW, why are there notes asking to verify if `make gateware` took less than 15 and building linux took less than 5 minutes anyway?

mithro, 2020-02-27:
@mateusz-holenko These were originally instructions to set for a day long tutorial. These constraints where to make sure people brought along laptops which where fast enough to get things done during the day.

mateusz-holenko, 2020-03-02:
@mithro I see - do you think it's still relevant for the version published on buildenv's wiki? 

What I mean is that timing is probably not that important for someone doing the tutorial at home + it's not quite clear what to do in case the times are above the limit - does it mean there is some kind of error, etc.

ewenmcneill, 2020-03-02:
> What I mean is that timing is probably not that important for someone doing the tutorial at home

Random thought: possibly the text could be changed to say something like "if this takes more than N minutes, you're going to spend a lot of time 'compiling' which might be frustrating and cause you take more than a day to get through the whole tutorial".  Then people playing along at home can decide whether or not they want to embark on that potentially painful route, or try to find a faster computer.  (AFAICR it's single core CPU (turboboost) speed that matters most to Vivado.)

As @mithro says, the original warning was there because the tutorial was originally run in a 6 hour "miniconf' slot, and if someone is waiting N * 10-15 minutes to even find out if they made the right change, 6 hours goes past pretty quickly  :-)   But I agree for someone at home they can always decide "I'll do something else while it's compiling" and carry on anyway.

Ewen

mateusz-holenko, 2020-03-04:
> Thanks @mithro -I removed all references to the text in red by rephrasing some fragments of the tutorial (in both versions).

I've just realized I made the changes on the antmicro's fork, not the upstream :(

It's fixed now - I synced both forks.

mateusz-holenko, 2020-03-04:
@ewenmcneill - Thanks for the suggestion! I allowed myself to include the additional note explainig why verifying of the compilation time might be important.

I guess this concludes the issue.

closed 2020-03-04 by mateusz-holenko