timvideos/litex-buildenv #323: PPA error for timsvideos

pdp7, 2020-02-05:
After doing the [LCA tutorial](https://github.com/timvideos/litex-buildenv/wiki/HowTo-LCA2018-FPGA-Miniconf-VexRiscv-Renode), I got this error when running apt update:
```
E: The repository
'http://ppa.launchpad.net/timvideos/fpga-support/ubuntu eoan Release'
does not have a Release file.
N: Updating from such a repository can't be done securely, and is
therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user
configuration details.
```

CarlFK, 2020-02-05:
> Not sure why that tutorial left you with that PPA enabled, but copied
> forward vizzini and flterm, to eoan, and focal.

thanks.

I'm not sure either, but I figured it was easy enough to bump that stuff
around and maybe it fixes it.
I'll test the tut .. prolly tomorrow.

pdp7, 2020-02-05:
Thanks.  I was working through the tutorial with an Arty board from @mithro at FOSDEM.  He recommended I log an issue.

mithro, 2020-02-06:
@antmicro are working on automated testing of the tutorials!
See;
 - https://github.com/antmicro/litex-linux-tutorial-pipeline
 - https://github.com/antmicro/litex-micropython-tutorial-pipeline
 - https://github.com/antmicro/litex-zephyr-tutorial-pipeline