timvideos/litex-buildenv #341: tftp: boot.bin not found

pdp7, 2020-02-19:
from tutorial: [HowTo-LCA2018-FPGA-Miniconf-VexRiscv-Renode](https://github.com/timvideos/litex-buildenv/wiki/HowTo-LCA2018-FPGA-Miniconf-VexRiscv-Renode#boot-linux-via-tftp)

I am trying the section "Boot Linux via tftp".  This [gist has the full log](https://gist.github.com/pdp7/287a0a5159d1877480aaaf6e513c23b1).

```
Feb 19 23:19:58 x1 atftpd[5324.140274570643200]: Serving boot.bin to 192.168.100.50:7642
Feb 19 23:19:58 x1 atftpd[5324.140274570643200]: File build/tftpd//boot.bin not found
Feb 19 23:19:58 x1 atftpd[5324.140274570643200]: Server thread exiting
```

pdp7, 2020-02-19:
this is reference I find to boot.bin:
```(LX P=arty C=vexriscv.linux F=linux) pdp7@x1:~/dev/litex-buildenv$ git grep boot.bin
Makefile:       cp $(FIRMWARE_FILEBASE).bin $(TFTPD_DIR)/boot.bin
```

pdp7, 2020-02-19:
```
(LX P=arty C=vexriscv.linux F=linux) pdp7@x1:~/dev/litex-buildenv$ ls -latr build/tftpd/
total 2488
drwxr-xr-x 7 pdp7 pdp7    4096 Feb 19 23:35 ..
-rw-r--r-- 1 pdp7 pdp7       0 Feb 19 23:35 rootfs.cpio
-rwxr-xr-x 1 pdp7 pdp7 2717924 Feb 19 23:35 Image
-rw-r--r-- 1 pdp7 pdp7    1842 Feb 19 23:35 rv32.dtb
-rwxr-xr-x 1 pdp7 pdp7    2992 Feb 19 23:35 emulator.bin
```

pdp7, 2020-02-19:
```
(LX P=arty C=vexriscv.linux F=linux) pdp7@x1:~/dev/litex-buildenv$ make firmware-connect
flterm --port=/dev/ttyUSB2 --speed=115200
[FLTERM] v2.4-29-g47d3b15 Starting...

litex> 
litex> reboot

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|

 (c) Copyright 2012-2019 Enjoy-Digital

 BIOS built on Feb 19 2020 23:27:55
 BIOS CRC passed (93665d52)

 Migen git sha1: d11565a
 LiteX git sha1: 02bfda5e

--=============== SoC ==================--
CPU:       VexRiscv @ 100MHz
ROM:       64KB
SRAM:      32KB
L2:        8KB
MAIN-RAM:  262144KB

--========== Initialization ============--
Ethernet init...
Initializing SDRAM...
SDRAM now under software control
Read leveling:
m0, b0: |11111111110000000000000000000000| delays: 05+-05
m0, b1: |00000000000000011111111111100000| delays: 21+-06
m0, b2: |00000000000000000000000000000001| delays: 31+-00
m0, b3: |00000000000000000000000000000000| delays: -
m0, b4: |00000000000000000000000000000000| delays: -
m0, b5: |00000000000000000000000000000000| delays: -
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |00000000000000000000000000000000| delays: -
best: m0, b1 delays: 21+-06
m1, b0: |11111111110000000000000000000000| delays: 05+-05
m1, b1: |00000000000000011111111111000000| delays: 20+-05
m1, b2: |00000000000000000000000000000001| delays: 31+-00
m1, b3: |00000000000000000000000000000000| delays: -
m1, b4: |00000000000000000000000000000000| delays: -
m1, b5: |00000000000000000000000000000000| delays: -
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |00000000000000000000000000000000| delays: -
best: m1, b1 delays: 20+-05
SDRAM now under hardware control
Memtest OK

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Loading emulator.bin from flash...
Error: Invalid image length 0xffffffff
Booting from flash...
Error: Invalid image length 0xffffffff
Booting from network...
Local IP : 192.168.100.50
Remote IP: 192.168.100.100
Fetching from: UDP/6069
Feb 19 23:40:31 x1 atftpd[18125.139949813991168]: socket may listen on any address, including broadcast
Feb 19 23:40:31 x1 atftpd[18125.139949813991168]: Serving Image to 192.168.100.50:7642
Feb 19 23:40:36 x1 atftpd[18125.139949813991168]: Server thread exiting
Feb 19 23:40:36 x1 atftpd[18125.139949805598464]: socket may listen on any address, including broadcast
Feb 19 23:40:36 x1 atftpd[18125.139949805598464]: Serving rootfs.cpio to 192.168.100.50:7642
Feb 19 23:40:36 x1 atftpd[18125.139949805598464]: Server thread exiting
Feb 19 23:40:36 x1 atftpd[18125.139949805598464]: socket may listen on any address, including broadcast
Feb 19 23:40:36 x1 atftpd[18125.139949805598464]: Serving boot.bin to 192.168.100.50:7642
Feb 19 23:40:36 x1 atftpd[18125.139949805598464]: File build/tftpd//boot.bin not found
Feb 19 23:40:36 x1 atftpd[18125.139949805598464]: Server thread exiting
Downloaded 2717924 bytes from Image over TFTP to 0xc0000000
Unable to download rootfs.cpio over TFTP
No rootfs.cpio found
Unable to download Linux images, falling back to boot.bin
Unable to download boot.bin over TFTP
Network boot failed
No boot medium found

--============= Console ================--
litex> 
```

pdp7, 2020-02-19:
**Note: rootfs.cpio is 0**
```
pdp7@x1:~/dev/litex-buildenv$ ls -tlar ./build/tftpd/
total 2488
drwxr-xr-x 7 pdp7 pdp7    4096 Feb 19 23:35 ..
-rw-r--r-- 1 pdp7 pdp7       0 Feb 19 23:35 rootfs.cpio
-rwxr-xr-x 1 pdp7 pdp7 2717924 Feb 19 23:35 Image
-rw-r--r-- 1 pdp7 pdp7    1842 Feb 19 23:35 rv32.dtb
-rwxr-xr-x 1 pdp7 pdp7    2992 Feb 19 23:35 emulator.bin
drwxr-xr-x 2 pdp7 pdp7    4096 Feb 19 23:35 .
pdp7@x1:~/dev/litex-buildenv$ ls -ltar ./build/arty_net_vexriscv.linux/software/linux/
total 20692
-rw-r--r--  1 pdp7 pdp7    1842 Jul 26  2019 rv32.dtb
drwxr-xr-x 10 pdp7 pdp7    4096 Feb  2 17:05 ..
-rw-r--r--  1 pdp7 pdp7       0 Feb  2 17:05 riscv32-rootfs.cpio
drwxr-xr-x  4 pdp7 pdp7    4096 Feb  2 17:05 include
drwxr-xr-x  3 pdp7 pdp7    4096 Feb  2 17:05 arch
drwxr-xr-x  2 pdp7 pdp7    4096 Feb  2 17:06 .tmp_versions
drwxr-xr-x  6 pdp7 pdp7    4096 Feb  2 17:06 scripts
drwxr-xr-x  2 pdp7 pdp7    4096 Feb  2 17:06 usr
drwxr-xr-x  2 pdp7 pdp7    4096 Feb  2 17:06 certs
drwxr-xr-x  2 pdp7 pdp7    4096 Feb  2 17:06 init
drwxr-xr-x  2 pdp7 pdp7    4096 Feb  2 17:06 ipc
drwxr-xr-x  2 pdp7 pdp7    4096 Feb  2 17:06 security
drwxr-xr-x  2 pdp7 pdp7    4096 Feb  2 17:06 crypto
drwxr-xr-x  3 pdp7 pdp7    4096 Feb  2 17:06 block
drwxr-xr-x  2 pdp7 pdp7    4096 Feb  2 17:06 sound
drwxr-xr-x  2 pdp7 pdp7    4096 Feb  2 17:06 firmware
drwxr-xr-x  2 pdp7 pdp7    4096 Feb  2 17:06 net
drwxr-xr-x  2 pdp7 pdp7    4096 Feb  2 17:06 mm
drwxr-xr-x  3 pdp7 pdp7    4096 Feb  2 17:06 virt
drwxr-xr-x 11 pdp7 pdp7    4096 Feb  2 17:06 fs
drwxr-xr-x 11 pdp7 pdp7    4096 Feb  2 17:06 kernel
drwxr-xr-x  6 pdp7 pdp7   12288 Feb  2 17:06 lib
drwxr-xr-x 42 pdp7 pdp7    4096 Feb  2 17:06 drivers
-rw-r--r--  1 pdp7 pdp7       2 Feb  2 17:06 .version
-rw-r--r--  1 pdp7 pdp7  176108 Feb  2 17:06 built-in.a
-rw-r--r--  1 pdp7 pdp7 6121292 Feb  2 17:06 vmlinux.o
-rw-r--r--  1 pdp7 pdp7       0 Feb  2 17:06 Module.symvers
-rwxr-xr-x  1 pdp7 pdp7 3016688 Feb  2 17:06 .tmp_vmlinux1
-rw-r--r--  1 pdp7 pdp7  811924 Feb  2 17:06 .tmp_kallsyms1.S
-rw-r--r--  1 pdp7 pdp7  140564 Feb  2 17:06 .tmp_kallsyms1.o
-rwxr-xr-x  1 pdp7 pdp7 3156196 Feb  2 17:06 .tmp_vmlinux2
-rw-r--r--  1 pdp7 pdp7  811924 Feb  2 17:06 .tmp_kallsyms2.S
-rw-r--r--  1 pdp7 pdp7  140564 Feb  2 17:06 .tmp_kallsyms2.o
-rwxr-xr-x  1 pdp7 pdp7 3156196 Feb  2 17:06 vmlinux
-rw-r--r--  1 pdp7 pdp7  371777 Feb  2 17:06 System.map
-rw-r--r--  1 pdp7 pdp7  371777 Feb  2 17:06 .tmp_System.map
-rw-r--r--  1 pdp7 pdp7     155 Feb  2 17:06 .vmlinux.cmd
-rw-r--r--  1 pdp7 pdp7 2717932 Feb 19 22:33 firmware.fbi
-rw-r--r--  1 pdp7 pdp7   22648 Feb 19 23:34 .config.old
-rw-r--r--  1 pdp7 pdp7   22648 Feb 19 23:34 .config
lrwxrwxrwx  1 pdp7 pdp7      47 Feb 19 23:34 source -> /home/pdp7/dev/litex-buildenv/third_party/linux
-rw-r--r--  1 pdp7 pdp7     437 Feb 19 23:34 Makefile
-rw-r--r--  1 pdp7 pdp7       0 Feb 19 23:34 modules.order
-rw-r--r--  1 pdp7 pdp7    1186 Feb 19 23:34 .missing-syscalls.d
lrwxrwxrwx  1 pdp7 pdp7      97 Feb 19 23:34 firmware.bin -> /home/pdp7/dev/litex-buildenv/build/arty_net_vexriscv.linux//software/linux/arch/riscv/boot/Image
drwxr-xr-x 22 pdp7 pdp7    4096 Feb 19 23:34 .
```

pdp7, 2020-02-19:
@mithro i believe the issue is that https://antmicro.com/projects/renode/litex-buildenv/ is `403 Forbidden`
```
pdp7@x1:~/dev/litex-buildenv$ wget https://antmicro.com/projects/renode/litex-buildenv/
--2020-02-19 23:48:59--  https://antmicro.com/projects/renode/litex-buildenv/
Resolving antmicro.com (antmicro.com)... 151.80.152.88
Connecting to antmicro.com (antmicro.com)|151.80.152.88|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2020-02-19 23:48:59 ERROR 403: Forbidden.
```

pdp7, 2020-02-19:
from https://github.com/timvideos/litex-buildenv/blob/master/scripts/build-linux.sh#L70
```
	export ARCH=riscv
	ROOTFS_LOCATION="https://antmicro.com/projects/renode/litex-buildenv/"
	ROOTFS=${ARCH}32-rootfs.cpio
```
and later:
```
                if [ ! -e $TARGET_LINUX_BUILD_DIR/$ROOTFS ]; then
                        wget $ROOTFS_LOCATION/$ROOTFS -O $TARGET_LINUX_BUILD_DIR/$ROOTFS
                fi
```

This turns out to be:
$TARGET_LINUX_BUILD_DIR = 
/home/pdp7/dev/litex-buildenv/build/arty_net_vexriscv.linux//software/linux

$TARGET_LINUX_BUILD_DIR/$ROOTFS =
/home/pdp7/dev/litex-buildenv/build/arty_net_vexriscv.linux//software/linux/riscv32-rootfs.cpio

$ROOTFS_LOCATION/$ROOTFS =
https://antmicro.com/projects/renode/litex-buildenv//riscv32-rootfs.cpio

```
(LX P=arty C=vexriscv.linux F=linux) pdp7@x1:~/dev/litex-buildenv$ wget https://antmicro.com/projects/renode/litex-buildenv//riscv32-rootfs.cpio
--2020-02-20 00:00:54--  https://antmicro.com/projects/renode/litex-buildenv//riscv32-rootfs.cpio
Resolving antmicro.com (antmicro.com)... 151.80.152.88
Connecting to antmicro.com (antmicro.com)|151.80.152.88|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4059136 (3.9M) [application/x-cpio]
Saving to: ‘riscv32-rootfs.cpio’

riscv32-rootfs.cpio                                  100%[=====================================================================================================================>]   3.87M  8.68MB/s    in 0.4s    

2020-02-20 00:00:55 (8.68 MB/s) - ‘riscv32-rootfs.cpio’ saved [4059136/4059136]
```

This does seem to work:
```
(LX P=arty C=vexriscv.linux F=linux) pdp7@x1:~/dev/litex-buildenv$ wget https://antmicro.com/projects/renode/litex-buildenv//riscv32-rootfs.cpio -O /home/pdp7/dev/litex-buildenv/build/arty_net_vexriscv.linux//software/linux/riscv32-rootfs.cpio
--2020-02-20 00:02:20--  https://antmicro.com/projects/renode/litex-buildenv//riscv32-rootfs.cpio
Resolving antmicro.com (antmicro.com)... 151.80.152.88
Connecting to antmicro.com (antmicro.com)|151.80.152.88|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4059136 (3.9M) [application/x-cpio]
Saving to: ‘/home/pdp7/dev/litex-buildenv/build/arty_net_vexriscv.linux//software/linux/riscv32-rootfs.cpio’

/home/pdp7/dev/litex-buildenv/build/arty_net_vexrisc 100%[=====================================================================================================================>]   3.87M  6.50MB/s    in 0.6s    

2020-02-20 00:02:21 (6.50 MB/s) - ‘/home/pdp7/dev/litex-buildenv/build/arty_net_vexriscv.linux//software/linux/riscv32-rootfs.cpio’ saved [4059136/4059136]
```

pdp7, 2020-02-19:
Now running again and rootfs.cpio is no longer zero:
```
(LX P=arty C=vexriscv.linux F=linux) pdp7@x1:~/dev/litex-buildenv$ make tftp

(LX P=arty C=vexriscv.linux F=linux) pdp7@x1:~/dev/litex-buildenv$ ls -ltar build/tftpd/
total 6452
drwxr-xr-x 7 pdp7 pdp7    4096 Feb 20 00:03 ..
-rwxr-xr-x 1 pdp7 pdp7 2717924 Feb 20 00:03 Image
-rw-r--r-- 1 pdp7 pdp7 4059136 Feb 20 00:03 rootfs.cpio
-rw-r--r-- 1 pdp7 pdp7    1842 Feb 20 00:03 rv32.dtb
-rwxr-xr-x 1 pdp7 pdp7    2992 Feb 20 00:03 emulator.bin
drwxr-xr-x 2 pdp7 pdp7    4096 Feb 20 00:03 .
```

pdp7, 2020-02-19:
Now it boots ok:
```
make gateware-load && make firmware-connect
```

Log:
```
(LX P=arty C=vexriscv.linux F=linux) pdp7@x1:~/dev/litex-buildenv$ make gateware-load && make firmware-connect
openocd -f board/digilent_arty.cfg -c "init; pld load 0 build/arty_net_vexriscv.linux//gateware/top.bit; exit"
Open On-Chip Debugger 0.10.0+dev-01012-ged8fa09cf-dirty (2020-01-15-04:14)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
jtagspi_program
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
flterm --port=/dev/ttyUSB2 --speed=115200
[FLTERM] v2.4-29-g47d3b15 Starting...
Initializing SDRAM...
SDRAM now under software control
Read leveling:
m0, b0: |11111111110000000000000000000000| delays: 05+-05
m0, b1: |00000000000000011111111111100000| delays: 21+-06
m0, b2: |00000000000000000000000000000001| delays: 31+-00
m0, b3: |00000000000000000000000000000000| delays: -
m0, b4: |00000000000000000000000000000000| delays: -
m0, b5: |00000000000000000000000000000000| delays: -
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |00000000000000000000000000000000| delays: -
best: m0, b1 delays: 21+-06
m1, b0: |11111111110000000000000000000000| delays: 05+-05
m1, b1: |00000000000000011111111111000000| delays: 20+-05
m1, b2: |00000000000000000000000000000001| delays: 31+-00
m1, b3: |00000000000000000000000000000000| delays: -
m1, b4: |00000000000000000000000000000000| delays: -
m1, b5: |00000000000000000000000000000000| delays: -
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |00000000000000000000000000000000| delays: -
best: m1, b1 delays: 20+-05
SDRAM now under hardware control
Memtest OK

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Loading emulator.bin from flash...
Error: Invalid image length 0xffffffff
Booting from flash...
Error: Invalid image length 0xffffffff
Booting from network...
Local IP : 192.168.100.50
Remote IP: 192.168.100.100
Fetching from: UDP/6069
Feb 20 00:08:11 x1 atftpd[18125.139949805598464]: socket may listen on any address, including broadcast
Feb 20 00:08:11 x1 atftpd[18125.139949805598464]: Serving Image to 192.168.100.50:7642
Feb 20 00:08:15 x1 atftpd[18125.139949805598464]: Server thread exiting
Feb 20 00:08:15 x1 atftpd[18125.139949813991168]: socket may listen on any address, including broadcast
Feb 20 00:08:15 x1 atftpd[18125.139949813991168]: Serving rootfs.cpio to 192.168.100.50:7642
Downloaded 2717924 bytes from Image over TFTP to 0xc0000000
Feb 20 00:08:22 x1 atftpd[18125.139949813991168]: Server thread exiting
Feb 20 00:08:22 x1 atftpd[18125.139949805598464]: socket may listen on any address, including broadcast
Feb 20 00:08:22 x1 atftpd[18125.139949805598464]: Serving rv32.dtb to 192.168.100.50:7642
Feb 20 00:08:22 x1 atftpd[18125.139949805598464]: Server thread exiting
Feb 20 00:08:22 x1 atftpd[18125.139949813991168]: socket may listen on any address, including broadcast
Feb 20 00:08:22 x1 atftpd[18125.139949813991168]: Serving emulator.bin to 192.168.100.50:7642
DownloFeb 20 00:08:22 x1 atftpd[18125.139949813991168]: Server thread exiting
[... 68 lines cut ...]
processor	: 0
hart		: 0
isa		: rv32ima
mmu		: sv32
uarch		: spinalhdl,vexriscv

# 
# uname -a
Linux buildroot 5.0.0 #1 Sun Feb 2 17:06:01 CET 2020 riscv32 GNU/Linux
# 
# dmesg |head
No DTB passed to the kernel
Linux version 5.0.0 (pdp7@x1) (gcc version 9.1.0 (9.1.0)) #1 Sun Feb 2 17:06:01 CET 2020
Initial ramdisk at: 0x(ptrval) (8388608 bytes)
Zone ranges:
  Normal   [mem 0x00000000c0000000-0x00000000c1ffffff]
Movable zone start for each node
Early memory node ranges
  node   0: [mem 0x00000000c0000000-0x00000000c1ffffff]
Initmem setup node 0 [mem 0x00000000c0000000-0x00000000c1ffffff]
On node 0 totalpages: 8192
# 
# 

````

mateusz-holenko, 2020-02-20:
That's a quite nice analysis of the problem - @pdp7 :)

It looks like some initial `wget` failed (network problem?) and it created an empty `riscv32-rootfs.cpio` file. From this moment on the (broken, empty) cached version was served by the tftp server.

The trivial solution would be to not only check if the rootfs file exists, but also verify it's not empty.
I sense more problems though - what if the file is only partially downloaded (i.e., it's not a proper rootfs image but it's not empty)?

We should probably add some CRC checking to make sure that files downloaded with `wget` are correct.

BTW: I also noticed a problem in `build-linux.sh` script in your log: https://gist.github.com/pdp7/287a0a5159d1877480aaaf6e513c23b1#file-litex-tftp-linux-txt-L3011. It's not related to your issue, but should be fixed anyway.

pdp7, 2020-02-20:
@mateusz-holenko thanks, that makes sense, so I could have also just deleted the empty rootfs.cpio as a workaround.

mateusz-holenko, 2020-03-04:
#347 added checksum verification of downloaded files, so this problem should not happen again.

closed 2020-03-04 by mateusz-holenko