timvideos/litex-buildenv #346: Linux on Renode = crash after 2nd char

CarlFK, 2020-02-22:
logged in as root, typed cat, I see the c, but it seems to be hung.

https://github.com/timvideos/litex-buildenv/wiki/HowTo-LCA2018-FPGA-Miniconf-VexRiscv-Renode#run-linux-in-renode

[image]

CarlFK, 2020-02-22:
let it sit, 8 hours later, it is unlockedup:

# cat /proc/cpuinfo 
processor : 0
hart  : 0
isa  : rv32ima
mmu  : sv32
uarch  : spinalhdl,vexriscv

mithro, 2020-02-23:
@mateusz-holenko

mateusz-holenko, 2020-02-25:
Thanks for reporting the issue, @CarlFK!

I tried to reporduce it locally, but (un)fortunately WFM.

I'm working on a CI setup to test linux on buildenv automatically in Travis - maybe this'll allow to catch some bugs.