timvideos/litex-buildenv #382: Modernize the PLL generation inside CRG files

mithro, 2020-04-06:
LiteX has gained the ability to easily create PLL objects. See https://github.com/timvideos/litex-buildenv/blob/master/targets/arty/crg.py for example.

Convert all the PLL generation in crg.py files to use this new method.

enjoy-digital, 2020-04-06:
Just for info, all up to date design are using this on Xilinx 7-series/Ultrascale(+) and Lattice iCE40/ECP5. The PLL object is also available on Spartan6 but has only been  tested with simple clocking (ie for SDRAM, it still needs to be adapted for the Pipistrello for example: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/pipistrello.py, it's not done mostly because i was not having the hardware). The Altera ALTPLL object is not yet done but i was planning to work on that soon.

enjoy-digital, 2020-04-08:
We now also have PLL objects for Altera/Intel chip and this has been integrated in LiteX-Boards, for example: https://github.com/litex-hub/litex-boards/commit/1fac6077fb9ebfb3c016e744b6e9279d799fceb5.