timvideos/litex-buildenv #384: Cleanup (and fix) the ice40_hx8k_b_env target

mateusz-holenko, 2020-04-07:
There are some debug leftovers in the target file: https://github.com/timvideos/litex-buildenv/blob/master/targets/ice40_hx8k_b_evn/base.py#L82.

This one needs testing on HW and (probably) some more fixing.

ewenmcneill, 2020-04-07:
See also https://github.com/timvideos/litex-buildenv/issues/277#issuecomment-610636356; I've just asked @niklasnisbeth to retest the HX8K EVB (that issue was the first report of what turned out to be litex memory size compatibility issues which are now fixed by #348).

And also https://github.com/timvideos/litex-buildenv/issues/304 is another report (from January) of issues with HX8K so they might be willing to test as well.  (It looks like that issue branched off into simulation issues, but the original report was HX8K EVB not booting.)

FTR, I don't have a iCE40 HX8K EVB myself.

Ewen

ewenmcneill, 2020-04-07:
See also https://github.com/timvideos/litex-buildenv/pull/313, which is an older WIP for the HX8K EVN board.  It'd probably need rebasing, but might have some helpful cleanup changes.

Ewen