timvideos/litex-buildenv #387: Make Linux image as good linux-on-litex-vexriscv

mithro, 2020-04-08:
Apparently people like the colored `buildroot#` prompt? See the discussion below;
> <timo.callahan> acomodi: Thanks, it worked! This Linux/sh  is nicer than the litex-buildenv version that I loaded over internet --- color "buildroot# " prompt rather than just a "# " prompt.
> <acomodi> @timo.callahan Great to hear that. Yeah, ethernet loading is definitely the best thing to use. And yeah, looks better with the colored `buildroot#` prompt :)

The linux root fs and other linux driver support in litex-buildenv should match the support in https://github.com/litex-hub/linux-on-litex-vexriscv

mithro, 2020-04-08:
FYI - @tcal-x @acomodi

tcal-x, 2020-04-08:
One other thing I noticed, and I don't know what part of the overall system it's due to (lxterm, serial port driver, etc.), but I noticed that when this litex-buildenv kernel was running, the first character that I typed was often dropped or corrupted.   This was more likely when I typed faster.   For example, if I typed "ls" over the serial connection, the board could get "s" or "xs".

When running the litex-hub gateware, the connection seemed more solid.

tcal-x, 2020-04-09:
You all know this I'm assuming, so I'm writing this down for my future reference.  
On https://github.com/timvideos/litex-buildenv/wiki/Linux, Tim mentions: 

> vexriscv
> - using the upstream kernel (without any litex-specific drivers, UART handled by VexRiscv's emulator)
> - using a patched buildroot 5.0.13 kernel from linux-on-litex-vexriscv (with litex-specific drivers)

With the two versions that I ran, the "basic" Linux that I built returned this from "uname -a":
`Linux buildroot 5.0.0 #1 Tue Apr 7 18:05:37 PDT 2020 riscv32 GNU/Linux`

while the "fancier" Linux returned this:
`Linux buildroot 5.0.13 #1 Tue Dec 3 08:55:41 CET 2019 riscv32 GNU/Linux`

mateusz-holenko, 2020-04-09:
> The linux root fs and other linux driver support in litex-buildenv should match the support in https://github.com/litex-hub/linux-on-litex-vexriscv

As @tcal-x mentioned, there already is an option in litex-buildenv to use binaries generated by https://github.com/litex-hub/linux-on-litex-vexriscv - it is not a default, though.
In order to do that you must set `BUILD_BUILDROOT` variable (https://github.com/timvideos/litex-buildenv/blob/master/scripts/build-linux.sh#L230) when running `./scripts/build-linux.sh` script.

> One other thing I noticed, and I don't know what part of the overall system it's due to (lxterm, serial port driver, etc.), but I noticed that when this litex-buildenv kernel was running, the first [...]

This is caused by the fact that the default Linux image does not come with any LiteX-specific drivers and UART is handled by the VexRiscv emulator. We also observed on  HW that this is not very reliable.

The linux-on-litex-vexrisv-based image comes with a LiteUART driver and this is much more stable.

mateusz-holenko, 2020-04-22:
The https://github.com/litex-hub/linux repository is slowly becoming the linux source for all LiteX projects, so I plan to test using this in litex-buildenv as well (both for VexRiscv and mor1kx).

mithro, 2020-04-23:
Sounds good to me!

mateusz-holenko, 2020-06-02:
Created a PR: https://github.com/timvideos/litex-buildenv/pull/450, but it must wait with merging for LiteX submodule to be upated first.

mateusz-holenko, 2020-07-14:
PR #450 is now merged to master. I guess this resolves the issue, what do you think @mithro / @tcal-x ?

mithro, 2020-07-15:
Probably! - @tcal-x can have the final say and close the bug.

tcal-x, 2020-07-15:
I have tested the most recent gateware + Linux firmware as described at https://github.com/timvideos/litex-buildenv/wiki/Linux.  

I encountered issue #483 , which was quickly resolved.

I loaded the Linux image using the tftp instructions, and logged in as root.   The prompt is indeed now colored, and the tty connection seems very solid (I saw no dropped or corrupted characters).

Only one very minor issue: when I first connected using `make firmware-connect`, I would see an extra LF (or CR?) after every command I typed, so that I'd always get two copies of the Linux prompt.  However, when I reconnected using `lxterm` directly, I did not see this behavior.

tcal-x, 2020-07-15:
Please close.   I do not have permission to close.

closed 2020-07-16 by mateusz-holenko