timvideos/litex-buildenv #426: How to build bare metal firmwares?

DurandA, 2020-05-06:
I am in the process of creating a bare metal firmware for my LiteX SoC. Since [the documentation](https://github.com/timvideos/litex-buildenv/wiki/Bare-Metal) is not yet ready (I am willing to help write this section once I figured how it works), I need some help.

As a first step, I wanted to build the [stub](https://github.com/timvideos/litex-buildenv/tree/master/firmware/stub) firmware so a copied the stub folder to a firmware project and used called `builder.add_software_package("firmware", src_dir=src_str)` before calling `builder.build`.

As [`STUB_DIRECTORY`](https://github.com/timvideos/litex-buildenv/blob/a3d3477a78ddfc856e87d5a3753f4baf8414cdf2/firmware/stub/Makefile#L36) was not defined, I replaced it with `FIRMWARE_DIRECTORY`. However, it seems that there are some issues with the includes:
```
make: Entering directory '/home/duranda/devel/liteboot/soc_basesoc_ecp5_evn/software/firmware'
 CC       main.o
/home/duranda/devel/liteboot/firmware/main.c: In function 'boot':
/home/duranda/devel/liteboot/firmware/main.c:16:2: error: implicit declaration of function 'irq_setmask' [-Werror=implicit-function-declaration]
  irq_setmask(0);
  ^~~~~~~~~~~
/home/duranda/devel/liteboot/firmware/main.c:17:2: error: implicit declaration of function 'irq_setie' [-Werror=implicit-function-declaration]
  irq_setie(0);
  ^~~~~~~~~
/home/duranda/devel/liteboot/firmware/main.c:18:2: error: implicit declaration of function 'flush_cpu_icache' [-Werror=implicit-function-declaration]
  flush_cpu_icache();
  ^~~~~~~~~~~~~~~~
```

* What is the normal process to build the firmware from the SoC? I used `builder.add_software_package("firmware", src_dir=src_str)`.
* Are env variables from the Makefile (e.g. `SOC_DIRECTORY`, `FIRMWARE_DIRECTORY` or `CPUENDIANNESS`) assigned by the SoC builder? I cannot find any reference to these.
* What about building the firmware apart from the SoC? I found a [firmware](https://github.com/gregdavill/OrangeCrab/blob/master/gateware/litex_ddr3_test/firmware/Makefile) from OrangeCrab from @gregdavill but I can't find where `SOC_DIRECTORY` is defined.

I think @mateusz-holenko may have some answers as he is the author of the stub firmware. :wink:

Thanks for the help!

gregdavill, 2020-05-06:
I'm not sure that firmware in the OrangeCrab repo actually builds correctly. But I can setup a project to show how I've done this later today as an example, I think there are a couple ways to do it.

the env variables are set by LiteX in a file called `variables.mak`. This is placed in `build/software/include/generated`

gregdavill, 2020-05-06:
https://github.com/litex-hub/fpga_101/tree/master/lab004 is a minimal example of how to create a firmware that is loaded by the bios, which might give you some pointers about how the puzzle pieces fit together.

DurandA, 2020-05-06:
Hi @gregdavill and thanks for the answers. I am eagerly waiting for my OrangeCrabs to arrive. :satisfied:

These _fpga_101_ examples are really awesome. I was able to build the firmware without errors. The issues of `error: implicit declaration of function 'irq_setmask'` came from the fact that I was using the SERV CPU which apparently is not configured with interrupts. I switched to VexRiscv and I had to edit _linker.ld_ as my board doesn't have any external RAM.

Unfortunately, the execution seems to hang after uploading the kernel:
```
--=============== SoC ==================--
CPU:       VexRiscv @ 60MHz
ROM:       32KB
SRAM:      4KB

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading firmware/firmware.bin to 0x40000000 (8476 bytes)...
[LXTERM] Upload complete (0.1KB/s).
[LXTERM] Booting the device.
[LXTERM] Done.
Executing booted program at 0x40000000

--============= Liftoff! ===============--
```

I most likely misconfigured memory regions as 0x40000000 does not seem to be mapped:
```
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
IO Regions: (1)
io0                 : Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False
Bus Regions: (3)
rom                 : Origin: 0x00000000, Size: 0x00008000, Mode: R, Cached: True Linker: False
sram                : Origin: 0x01000000, Size: 0x00001000, Mode: RW, Cached: True Linker: False
csr                 : Origin: 0x82000000, Size: 0x00010000, Mode: RW, Cached: False Linker: False
```

gregdavill, 2020-05-06:
You might be mixing examples, you're right that linker script doesn't look right.

I've pushed a simple example of how I've been creating an embedded firmware. https://github.com/gregdavill/OrangeCrab-examples/tree/master/litex/SoC-baremetal

This is essentially replacing the bios with your ovn custom code, but it's also building litex libbase and compiler_rt. Which enables you to use things like printf and uart drivers that the standard bios uses.

Some of the code in there is specific to the ECP5 and working in `ecpbram` to enable updating of the firmware without having to re-syntheses the design. But hopefully that helps.

DurandA, 2020-05-08:
@gregdavill Thank you very much, that was very helpful.

I took some notes and will write a paragraph on the [wiki](https://github.com/timvideos/litex-buildenv/wiki/Bare-Metal) in the next few days so beginners won't fall in the same pitfalls than me (e.g. linking to the same memory regions than the bios :rofl:).

closed 2020-05-08 by DurandA