timvideos/litex-buildenv #438: LiteX RISCV + MicroPython on Mimas Spartan 6 dev board

dayjaby, 2020-05-21:
Hello :)

I have a Mimas Spartan 6 development board and I'm trying to run Micropython on it. For that, I added mimas as a platform and target (see https://github.com/dayjaby/litex-buildenv/commit/4d30cd62a47097c4c75a1a1a451d19309c00e2eb - mostly stuff copied from mimasv2.py). Everything seems to work fine, even uploading the firmware, but I am not able to connect to the board via flterm:

```
$ make firmware-load
riscv64-unknown-elf-ld:/home/david/Development/litex/litex-buildenv/firmware/stub/linker.ld:15: warning: memory region `user_flash' not declared
chmod -x firmware.elf
 OBJCOPY  firmware.bin
chmod -x firmware.bin
python -m litex.soc.software.mkmscimg -f --little firmware.bin -o firmware.fbi
make[1]: Leaving directory '/home/david/Development/litex/litex-buildenv/build/mimas_base_vexriscv/software/stub'

real	0m1,823s
user	0m1,533s
sys	0m0,248s
flterm --port=/dev/ttyACM0 --kernel=build/mimas_base_vexriscv/software/micropython/firmware.bin --speed=19200
[FLTERM] v2.4-29-g47d3b15 Starting...
```

And I also tried image-flash:
```
$ make image-flash
...
Flash image: build/mimas_base_vexriscv/image-gateware+bios+micropython.bin
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff aa 99 55 66 30 a1 00 07 20 00 31 a1 03 80 31 41 3d 00 31 61 09 ee 31 c2 04 00 10 93 30 e1 00 cf 30 c1 00 81 20 00 20 00 20 00 20 00 20 00 20 00
python -m MimasV2.Config /dev/ttyACM0 build/mimas_base_vexriscv/image-gateware+bios+micropython.bin
****************************************
* Numato Lab Mimas V2 Configuration Tool *
****************************************
Micron M25P16 SPI Flash detected
Loading file build/mimas_base_vexriscv/image-gateware+bios+micropython.bin...
Erasing flash sectors...
Writing to flash 100% complete...
Verifying flash contents...
Flash verification successful...
Booting FPGA...
Done.
```

However, I am not able to connect to the micropython firmware:
```
make firmware-connect
flterm --port=/dev/ttyACM0 --speed=19200
[FLTERM] v2.4-29-g47d3b15 Starting...
```

What step am I missing? I've read something about a WishboneUARTBridge; is it possible to connect to MicroPython via this bridge?

ewenmcneill, 2020-05-22:
If i'm understanding correctly, you have an [original Mimas](https://numato.com/product/mimas-spartan-6-fpga-development-board) rather than a [Mimas v2](https://numato.com/product/mimas-v2-spartan-6-fpga-development-board-with-ddr-sdram)?  

If so, one of the relevant differences is that the Mimas v2 has external DDR RAM ("DDR Memory: 166MHz 512Mb LPDDR"), and the original Mimas does not.  Litex being a "System on a Chip" system based around a Soft CPU pretty much assumes there'll be some RAM available for the Soft CPU to use.  (IIRC the Spartan 6 LX9 does have a small amount of block/distributed RAM, but from memory it's not enough to really start up a soft CPU on the LX9; some other FPGAs do have more "built in" RAM and have a chance of running a soft CPU without external DDR RAM.)

Another relevant difference is that the Mimas v2 has a serial interface chip (a PIC microcontroller IIRC) which has a switch to change between "upload to the SPI Flash to program the FPGA" and "connect to serial port on FPGA to see text from whatever is running".  It doesn't look like the Original Mimas has that feature either (ie, switching what the USB CDC serial is connected to), so the PIC microcontroller for USB serial might be limited to just programing the SPI Flash for the FPGA.  Which would mean you might need to use another set of FPGA pins which are broken out to the GPIO header as your serial console -- and use an external USB CDC serial adapter.

Both of those are possibly show stopper problems for using the Original Mimas board with Litex.  So you might want to consider doing something other than Litex with the FPGA dev board (eg, a "non soft CPU" based design), or looking at another small FPGA board.  (Ordinarily I'd suggest the iCEBreaker as a good "starter" candidate, but I believe they're almost out of stock right now -- CrowdSupply / Mouser might have some maybe?)

If you have some other board than that Original (cheaper) Mimas Spartan 6 board, perhaps you could be more precise about which board you're trying to get working?

Ewen

PS: I believe the WishboneUARTBridge is intended only for debugging (internal bus access), not for serial console.  And it would rely on having a working UART (eg, USB CDC serial, or external adapter) path -- which is the problem you started with -- since it connects to the internal System on a Chip inside the FPGA over a serial connection :-)

dayjaby, 2020-05-23:
Thank you so much for this elaborate answer! Yes, I got the "original" mimas. Would it be possible to create some RAM as part of the FPGA? 4Kb (or better: 8Kb) RAM seems to be sufficient according to https://forum.micropython.org/viewtopic.php?t=28#p48

For serial connection, I have a USB-FTDI available, with which I could connect well to the board according to https://github.com/litex-hub/fpga_101/blob/master/lab003/test/test_buttons_switches.py. 

Regarding other boards: I also have a Pynq Z2 available. Could this be used for doing experiments, e.g. running LiteX Vexriscv + micropython?

ewenmcneill, 2020-05-24:
It looks like you *might* be able to make it work on the Spartan 6 LX9, maybe, if the RAM storage was mapped to FPGA block RAM, with some address decoding to spread the RAM usage over several block RAMs -- as each block RAM is about 2kB.

But I'm not sure how much of that will happen automatically in litex/Xilinx's tools, and how much you'd have to build out of individual bits yourself.  It'd definitely be more difficult than on the Mimas v2 where there's a larger external RAM chip that is linearly addressable in one chunk.  And possibly not a "beginner" project.

FPGAs are made up of "logic blocks" which typically contain a LUT (Look Up Table), a flip-flop (or two; single bit registers), and occasionally some other features.   The LUTs are effectively read-only (only set at "programming" the FPGA time), but the flip-flops can store a (single bit) value.   But especially on a small FPGA, the flip flops won't store that much data -- the Spartan 6 LX9 seems to have 11,440 flip flops -- which works out to be 1430 bytes of storage total, or about 1.4kB, which is mostly used up by *implementing* the soft CPU (eg, CPU registers, nd other state, needs some storage space!), so it's clearly not the RAM you need.  (See the table in the [Spartan 6 Datasheet](https://www.xilinx.com/support/documentation/data_sheets/ds160.pdf) on page 2.)

To work around this limitation, most modern FPGAs have some special additional features built in, including "distributed RAM" (a little bit of RAM in each of several of the logic cells, usually used as part of the implementation of, eg, a soft CPU) and/or "block RAM" (special cells which just contain a small block of RAM) which is often used as more generic RAM.  The Spartan 6 LX9 has 90kbits (about 11kB)  of distributed RAM, and 32 * 18kbits (2250kB each) block RAM.  That's a reasonable amount of bock RAM -- but it's divided up into small 2kB chunks (and if you use *any* of a block, the whole block is usually not available for anything else).  Block RAM typically needs some custom interfacing logic to address it, etc, to "glue it together" into a a logical RAM block for a soft CPU.  As I said above, I don't know if litex can do that for you -- but it'd definitely be a different config from the Mimas v2, which is pointing the RAM at the external linear RAM chip.

The USB FTDI adapter you have should work for USB serial though.  If you map the soft CPU "serial" pins across to pins on the FPGA that go out to one of the external GPIO connector in an order that you can plug in your USB FTDI adapter.  (Beware that TX on one device needs to go to RX on the other so they "hear" each other :-) ).   You might be able to test that functionality by finding a design online that will just output serial information without a CPU, eg, continually outputting "Hello World" or similar, so you just had to get the pin mapping right.

Ewen

ewenmcneill, 2020-05-24:
I'm not familiar with the Pynq boards, but it looks like the [Pynq Z2](http://www.tul.com.tw/ProductsPYNQ-Z2.html) is [based on a Xilinx Zynq Z7020 FPGA](http://www.pynq.io/board.html).  The Zynqs are ARM CPUs (hard core) *and* FPGAs in one package, and typically the DDR RAM is attached to the ARM CPU, and the FPGA half is programmed from the ARM CPU side (ie the ARM CPU has to boot first).  It looks like there's a reasonable size FGPA module in it, and more block RAM (630KB? attached to the FPGA directly), so it could be the better option of the two to consider at least for "small RAM" uses (eg, like MicroPython, which is happy with 8kB-16kB of RAM as a minimum).

It looks like there's a bit of work on [supporting the Zynq range in upstream litex](https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc_zynq.py), but I'm not sure of the state of that.  I do know that because of the ARM CPU / RAM / FPGA combination there's been less work on supporting the Zynq style chips than, eg, bare Artix 7 FPGAs or Spartan 6, iCE40, ECP5, etc.  So it might be a bit of work to get litex going -- and you might need to ask directly in the litex IRC channel (`#litex` on FreeNode) or their GitHub project.  (litex-buildenv is basically an "install a build environment" for litex, with some board definitions of its own for historical reasons.)

Good luck!

Ewen

dayjaby, 2020-05-24:
Thank you Ewen! That answers all my questions (for now) :)

closed 2020-05-24 by dayjaby