timvideos/litex-buildenv #460: Make gateware fails for possible all ISE based Boards

cklarhorst, 2020-06-19:
Using the latest master branch it doesn't seem to be possible to build the gateware for any ISE based Board. The process fails with the error message:
```Writing file top_map.ngm...
Running directed packing...
ERROR:Pack:1107 - Pack was unable to combine the symbols listed below into a
   single IOB component because the site type selected is not compatible.
   Further explanation:
   The buffer "IBUFG" can not merge with signal "clk100." The signal has a
   NOMERGE property.
   Symbols involved:
        BUF symbol "IBUFG" (Output Signal = crg_clk100a)
        PAD symbol "clk100" (Pad Signal = clk100)
```
So the problem seems to be the keep property that is added to the clk100 signal in the top.v file. 
```
(* keep = "true" *)     input clk100,
```
Git bisect revealed that the last working commit is 688d49f4219de0fb79e8c8fafb3dd3c588905c20.
This version doesn't add the property to the mentioned signal.

Is it already a known problem?
Does someone know why the keep property was added? Is it safe to remove it?

Thank you very much in advance

mithro, 2020-06-23:
@mateusz-holenko - Any idea?

mateusz-holenko, 2020-06-25:
Hi @cklarhorst!

I can confirm it is also visible in CI: https://travis-ci.com/github/antmicro/litex-buildenv/jobs/346569019#L6459-L6467. I'm currently working on bumping the LiteX submodule in litex-buildenv, but it doesn't look it will solve *this* problem.

@enjoy-digital do you have any suggestions?

cklarhorst, 2020-06-25:
Sadly I'm no expert on that front but according to [ug901-vivado-synthesis.pdf](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2013_4/ug901-vivado-synthesis.pdf) page 46-47 KEEP=True in Vivado means: Keep the signal during synthesis and no constrains for the later stages like P&R. While according to the [Xilinx Constraints Guide](https://www.xilinx.com/support/documentation/sw_manuals/xilinx14_7/cgd.pdf) page 135 for ISE it means "Prevents a net from being absorbed into a logic block" but it also implies a NOMERGE constraint for the later implementation phase. 

Maybe for ISE, it is required to set KEEP to "soft" so that it doesn't imply the NOMERGE constraint. (A quick test showed that ISE successfully builds a bitstream with KEEP="soft").