timvideos/litex-buildenv #483: download-env.sh: Installing pythondata-software-compiler_rt (local python module): can't open file 'setup.py'

tcal-x, 2020-07-15:
Following instructions on https://github.com/timvideos/litex-buildenv/wiki/Linux, under "Building Linux", "Download and Configure the Build Environment", at the `./scripts/download-env.sh` step, things are going fine until:

```
...
...
Installing robotframework (python module)
Collecting robotframework==3.0.4
  Downloading robotframework-3.0.4.tar.gz (443 kB)
     |████████████████████████████████| 443 kB 1.1 MB/s 
Building wheels for collected packages: robotframework
  Building wheel for robotframework (setup.py) ... done
  Created wheel for robotframework: filename=robotframework-3.0.4-py3-none-any.whl size=585654 sha256=b15c9e6dc0cc680cfbbff2af5ef08cd5f8a25c0dbad73107e07bb3ee65703822
  Stored in directory: /home/tim/.cache/pip/wheels/f3/1b/1d/eeb97315a5faf8dbf7bc7afd85a8b91c42e5592a856b05771f
Successfully built robotframework
Installing collected packages: robotframework
Successfully installed robotframework-3.0.4
robot found
Installing pythondata-software-compiler_rt (local python module)
python: can't open file 'setup.py': [Errno 2] No such file or directory
```
I saw the same behavior on two systems:
* Ubuntu 18.04 (GCP)
* Ubuntu 19.04 (personal laptop)

Let me know if you want the full output from download-env.sh.

cklarhorst, 2020-07-15:
Please look into `third_party/pythondata-software-compiler_rt` if it's empty then most likely the submodules aren't initialized.
Just do `git submodule update --init --recursive` that will download all submodules.
Maybe the clone line in the wiki should be updated with a `--recurse-submodules`  flag?

mateusz-holenko, 2020-07-15:
I agree - it's probably a not initialized submodule. 

`./scripts/download-env.sh` should initialize and update all submodules, but from what I see it does it *after* installing python modules (https://github.com/timvideos/litex-buildenv/blob/master/scripts/download-env.sh#L697). It's a bug.

tcal-x, 2020-07-15:
* Yes, in the original case, doing my own `git submodule update --recursive` allowed me to get past the issue.

* I tested with Matt's fix and a fresh clone, and confirmed that it now works without doing an explicit submodule update.

Thanks for the quick fix!

closed 2020-07-15 by tcal-x