timvideos/litex-buildenv #485: Current version of make.py is incompatible with the litescope module

cklarhorst, 2020-07-15:
Adding the litescope module to a target file results in a failed `make gateware` because make.py:180 will call `do_exit` with `filename` as an argument.
```
Traceback (most recent call last):
  File "./make.py", line 185, in <module>
    main()
  File "./make.py", line 181, in main
    soc.do_exit(vns, filename="{}/analyzer.csv".format(testdir))
  File "[cut]/third_party/migen/migen/fhdl/module.py", line 185, in do_exit
    submodule.do_exit(*args, **kwargs)
TypeError: do_exit() got an unexpected keyword argument 'filename'
```
The problem is that the `do_exit` function in the litescope module doesn't support variable parameters.

To my understanding that is a bug in litescope. `do_exit` needs to accept  `*args, **kwargs`. Correct?

But then the question is why does the make.py in this repo call do_exit with filename as a parameter?
The LiteScopeAnalyzer already supports the configuration of the csv file in the constructor. 
So to my understanding make.py should not add the filename to the `do_exit` parameters.

mithro, 2020-07-15:
@mateusz-holenko - Thoughts?

cklarhorst, 2020-07-19:
Ok. I just found `targets/mimasv2/scope.py` and the idea seems to be that a target just overrides the default `do_exit `logic. I think this was done because there is no path information for the csv file at the time when the LiteScopeAnalyzer constructor is called. I will close this issue because I think that this solution is ok.

closed 2020-07-19 by cklarhorst