timvideos/litex-buildenv #616: Micropython build fails on WSL2

whatnick, 2021-01-06:
On WSL2 the micropython build script errors out with `too many input files` in the pre-processor stage. Works fine on native Ubuntu. 
[image]

mithro, 2021-01-06:
You are missing a lot of important information I need to figure out what is going on here.

What is the exact command you running on what commit revision? What were the lines before and after this one? What have you done to try and debug?

(BTW A screenshot is also a **terrible** way to put the output of something into a issue report.)

whatnick, 2021-01-08:
I had it saved. Better do it properly and contribute an issue template for this repo too. Here are details:

OS : Ubuntu on WSL2
Relevant ENV VARS:
PLATFORM : icebreaker
CPU : vexriscv
CPU_VARIANT : lite
TARGET : base
FIRMWARE : micropython

BUILD Stage: building firmware prior to make image for icepacked .bin file,  running on the head of this PR #614  

CLI Prefix : `(LX P=icebreaker C=vexriscv.lite F=micropython R=support-icefeather)`

Running : `./scripts/build-micropython.sh`
Crashes with :

```
fatal error : Too many input files
```

The same sequence of actions works fine on NON WSL2 Ubuntu. Some discussion with @shenki suggests issues with platform detection. Unfortunately I have not done much debugging other than complaining loudly about WSL2 vagaries.