timvideos/litex-buildenv #753: download-env.sh fails due to liteusb

lachlansmith, 2022-04-05:
Hi the download environment script never goes as far as installing the litex suite because it fails to clone liteusb. Is liteusb no longer a thing? I'm getting a 404 from https://github.com/enjoy-digital/liteusb.git/ is this just me?

```
Cloning into '/home/lachy/test/litex-buildenv/third_party/VexRiscv'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/edid-decode'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/flash_proxies'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/litedram'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/liteeth'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/liteiclink'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/litepcie'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/litesata'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/litescope'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/liteusb'...
Username for 'https://github.com': 
Password for 'https://github.com': 
remote: Repository not found.
fatal: Authentication failed for 'https://github.com/enjoy-digital/liteusb.git/'
fatal: clone of 'https://github.com/enjoy-digital/liteusb.git' into submodule path '/home/lachy/test/litex-buildenv/third_party/liteusb' failed
Failed to clone 'third_party/liteusb'. Retry scheduled
Cloning into '/home/lachy/test/litex-buildenv/third_party/litevideo'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/litex'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/litex-boards'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/litex-renode'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/migen'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/nmigen'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/pythondata-cpu-lm32'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/pythondata-cpu-minerva'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/pythondata-cpu-mor1kx'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/pythondata-cpu-picorv32'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/pythondata-cpu-vexriscv'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/pythondata-software-compiler_rt'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/valentyusb'...
Cloning into '/home/lachy/test/litex-buildenv/third_party/liteusb'...
Username for 'https://github.com': 
Password for 'https://github.com': 
remote: Repository not found.
fatal: Authentication failed for 'https://github.com/enjoy-digital/liteusb.git/'
fatal: clone of 'https://github.com/enjoy-digital/liteusb.git' into submodule path '/home/lachy/test/litex-buildenv/third_party/liteusb' failed
Failed to clone 'third_party/liteusb' a second time, aborting

```

mithro, 2022-05-21:
Should be fixed by https://github.com/timvideos/litex-buildenv/pull/754 - can people verify?

wohlbier, 2022-05-21:
I ran this `curl -fsS https://raw.githubusercontent.com/timvideos/litex-buildenv/master/scripts/bootstrap.sh | bash` and the `third_party` stuff seems to work, but then it fails as below. Looking at why.
```
Already on 'master'
Your branch is up to date with 'origin/master'.
             This script is: /home/jgwohlbier/devel/litex-buildenv/scripts/download-env.sh
         Firmware directory: /home/jgwohlbier/devel/litex-buildenv
         Build directory is: /home/jgwohlbier/devel/litex-buildenv/build
     3rd party directory is: /home/jgwohlbier/devel/litex-buildenv/third_party

Initializing environment
---------------------------------

Installing conda (self contained Python environment with binary package support)
2022-05-21 12:40:06 URL:https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh [76607678/76607678] -> "Miniconda3-latest-Linux-x86_64.sh" [1]
PREFIX=/home/jgwohlbier/devel/litex-buildenv/build/conda
Unpacking payload ...
./Miniconda3-latest-Linux-x86_64.sh: 412: /home/jgwohlbier/devel/litex-buildenv/build/conda/conda.exe: Exec format error
./Miniconda3-latest-Linux-x86_64.sh: 414: /home/jgwohlbier/devel/litex-buildenv/build/conda/conda.exe: Exec format error
```

wohlbier, 2022-05-21:
Ok, this has to do with me being on an M1. I get further with the following patch
```
jgwohlbier@m1vm:~/.../litex-buildenv$ git diff
diff --git a/scripts/download-env.sh b/scripts/download-env.sh
index bb77de0..baebf21 100755
--- a/scripts/download-env.sh
+++ b/scripts/download-env.sh
@@ -250,16 +250,17 @@ export PATH=$CONDA_DIR/bin:$PATH:/sbin
        echo "Installing conda (self contained Python environment with binary package support)"
        if [[ ! -e $CONDA_DIR/bin/conda ]]; then
                cd $BUILD_DIR
+               MINICONDA=Miniconda3-latest-Linux-aarch64.sh
                # FIXME: Get the miniconda people to add a "self check" mode
-               wget --no-verbose --continue https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh
+               wget --no-verbose --continue https://repo.anaconda.com/miniconda/${MINICONDA}
                #wget --continue https://repo.anaconda.com/miniconda/Miniconda3-${CONDA_VERSION}-Linux-x86_64.sh -O Miniconda3-latest-Linux-x86_64.sh
-               chmod a+x Miniconda3-latest-Linux-x86_64.sh
+               chmod a+x ${MINICONDA}
                # -p to specify the install location
                # -b to enable batch mode (no prompts)
                # -f to not return an error if the location specified by -p already exists
                (
                        export HOME=$CONDA_DIR
-                       ./Miniconda3-latest-Linux-x86_64.sh -p $CONDA_DIR -b -f || exit 1
+                       ./${MINICONDA} -p $CONDA_DIR -b -f || exit 1
                )
                fix_conda
                conda config --system --add envs_dirs $CONDA_DIR/envs
```

but then I get to
```
Installing FPGA toolchain
---------------------------------------

Installing yosys (FOSS Synthesis tool)
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - yosys

Current channels:

  - https://conda.anaconda.org/timvideos/linux-aarch64
  - https://conda.anaconda.org/timvideos/noarch
  - https://repo.anaconda.com/pkgs/main/linux-aarch64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/linux-aarch64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```

wohlbier, 2022-05-21:
I tested it on linux x86 and it installs fine.